aredn/aredn #1544: Tools menu doesn't close after Support Data extract

m0xzl, 2024-09-19:
### Describe the bug
Once a Support Data file has been downloaded via the 'Tools' menu, the pop-up menu window does not close.

### Expected behavior
Once the download is complete the pop-up menu should close.

### Screenshots
[image]

### Additional context
* When you return from using other facilities on the 'Tools' menu the pop-up menu is closed.

aanon4, 2024-09-19:
Oddly tricky to do as I dont want to close the menu until the download is done, and the download isnt easy for the front end to detect.

m0xzl, 2024-09-19:
Understood. Not really a problem, just an anomaly I spotted.

aanon4, 2024-09-19:
Spoke too soon ...
https://github.com/aredn/aredn/pull/1546

closed 2024-09-23 by aanon4