aredn/aredn #1545: Mesh Nodes statistic appears to be one short

m0xzl, 2024-09-19:
### Describe the bug
On the 'mesh' page I count 17 nodes. On the 'status' page it reports 16 nodes i.e. one less.

### Expected behavior
The 'nodes' statistic should match the number of nodes listed on the 'mesh' page.

### Screenshots
[image]
[image]

### Additional context
* Have I misunderstood what the 'nodes' statistic is intended to represent?

aanon4, 2024-09-19:
Could you attach the support data and confirm the above information at the time you download it (as these numbers can change)? The node count is generated by counting dtdlink entries in /var/run/hosts_olsr, and there should be one for each node. So I'm curious which node doesnt have one and why not.

m0xzl, 2024-09-19:
[supportdata-M0XZL-GL5-2024-9-19-8-31.tar.gz](https://github.com/user-attachments/files/17055278/supportdata-M0XZL-GL5-2024-9-19-8-31.tar.gz)

aanon4, 2024-09-19:
M0XZL-L1 appears to be different. Anything particular about it?

m0xzl, 2024-09-19:
Yes, M0XZL-L1 is an old Linksys node running BBHN. It has been modified to provide a simple version of the SysInfo API. I leave it on the network because it is sometimes good to have a comparison for debugging code that I write to monitor things about the nodes.

I believe it is M7SHF-GL1 that is missing from the hosts_olsr file.

aanon4, 2024-09-19:
M7SHF-GL1 is there.
[image]

aanon4, 2024-09-19:
So I guess the node count is correct, because one of the devices is not an AREDN node (in the way we count them)

m0xzl, 2024-09-19:
OK, I can cope with that now that I understand the difference.