aredn/aredn #1549: Max Distance doesn't match Old UI vs. New UI

k1ky, 2024-09-20:
### Describe the bug
A clear and concise description of what the bug is.
Maximum distance is different in new UI as compared to Old UI in some cases.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes - N-20240919

### Expected behavior
A clear and concise description of what you expected to happen.
Expect the Max Distance and any other values to match between UI's.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
[image]
New UI shows 50 miles
[image]
Old UI shows 1.24 miles.  Would also be nice for the new UI to show the decimal value to 2 places to match old UI

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-09-20:
You pesky people who turn LQM off.
https://github.com/aredn/aredn/pull/1550

aanon4, 2024-09-23:
Can you verify this has been fixed? Thanks.

k1ky, 2024-09-23:
Looking much better - I am able to make changes in either the new and old UI and the numbers somewhat track with LQM enabled and Disabled.

closed 2024-09-23 by k1ky