aredn/aredn #1565: WG tunnel diapered from the list on w6bbhub-4 when i modified the weight

mathisono, 2024-09-23:
WG tunnel diapered from the list on my main tunnel node w6bbhub-4, After modifying weight. 

Im now unable to add any new tunnel client enters that work.

Node appears to be unstable.

support data attached:

[supportdata-W6BB-RFS-HUB4-2024-9-22-13-17.tar.gz](https://github.com/user-attachments/files/17092780/supportdata-W6BB-RFS-HUB4-2024-9-22-13-17.tar.gz)
[supportdata-KJ6DZB-WSB-hub5-2024-9-22-13-30.tar.gz](https://github.com/user-attachments/files/17092781/supportdata-KJ6DZB-WSB-hub5-2024-9-22-13-30.tar.gz)
[supportdata-W6BB-RFS-HUB4-2024-9-22-13-30.tar.gz](https://github.com/user-attachments/files/17092783/supportdata-W6BB-RFS-HUB4-2024-9-22-13-30.tar.gz)

aanon4, 2024-09-23:
Which tunnel entry did you modify?
Was it a client or server tunnel?
What was the weight before the change?
What was the weight after the change?
Was any part of the tunnel highlighted in red when you entered done?
Do you recall how many changes pending listed before you hit commit?
--
What does "the node appears to be unstable" mean exactly? Is it crashing or something else?
Can you provide the details of the tunnel entry you attempted to add?
When it fails, how does it fail?
--
Can you tell me the state of the node for each of the included support datas and why one has a different node name?

mathisono, 2024-09-23:
Which tunnel entry did you modify?
hub-5

Was it a client or server tunnel?
server

What was the weight before the change?
1 changed to .5

What was the weight after the change?
.5

Was any part of the tunnel highlighted in red when you entered done?
No

Do you recall how many changes pending listed before you hit commit?
no clue

What does "the node appears to be unstable" mean exactly? Is it crashing or something else?
This is what now appears to be the LQM displaying information on the status page. Ive updated a new nodes with nighty this evening. Ive seen this at boot, the status page loads, olsrd has discovered and populated "Local Nodes/ Neighborhood Nodes"   but LQM is indicating the node a GRAY not GREEN as seen here in pic. The hover over indicates idle. This takes ~2 min to recover to green. Ive watched this on my on 3 nodes now. I noticed it on hub-4 in this bug because the tunnel for WSB-hub5 turned red after the change applied.
[image]

Can you provide the details of the tunnel entry you attempted to add?
I attempted to add a new wireguard tunnel on the server, the connection info was added input to the client and It never connected.

When it fails, how does it fail? The entry has diapered from the server! and the client no longer connects 
One and done , as soon as I made the change, ive never seen any thing connect again.

Can you tell me the state of the node for each of the included support datas and why one has a different node name?

aanon4, 2024-09-23:
> hub-5
I assume you mean kj6dzb-wsb-hub5

> server
Thanks

> .5
You will have seen the box border turn red when you entered a factional weight. Integers only. What then happened is you tried to save an illegal entry (the box was red) so the entry was not saved. Because we cannot store illegal data, the entry was removed when you ignored the red marker and committed your changes.

Now, I'm not going to claim this is a good thing because it isn't; so from tomorrow you will no longer be able to press 'Done' if any entry in the tunnels dialog is illegal. Hopefully this will prevent this happening to other people.

> No.
I think you're mistaken. Validation is done in the html markup and I also rechecked that .5 is correctly marked as invalid in all themes (I went through and made sure). Unless you're using a very old browser which doesnt support this? (but the feature is very old).

> No clue
I can hope.

The image you include is not from w6bbhub-4 and, from what you're describing, doesnt seem relevant to this issue.

> new tunnel entry
To be clear, you created a new wireguard tunnel server entry on w6bbhub-4 and attempt to connect to it by adding a wireguard tunnel client on hub5, is this correct?

> When it fails, how does it fail? The entry has diapered from the server! and the client no longer connects

Do you create a new entry on the hub-4, give it a name, hit done, see the commit, hit commit .. and it's no longer there at that point?

> One and done , as soon as I made the change, ive never seen any thing connect again.

Again, to clarify, when you create new server entry on hub4, once you commit, all wireguard connections to hub4 are terminated and never recover? I assume you waited a while for this (it does have to restart the entire network stack at that point); how long?

aanon4, 2024-09-23:
Just to test, I created a wireguard server on hub4 to connect to my home tunnel client, merlin. Connection worked just fine. I have now deleted it.

mathisono, 2024-09-23:
do you see the entry for wsb-hub5?

aanon4, 2024-09-23:
> do you see the entry for wsb-hub5?

No, I dont see an entry in the UI or the support data (last of the three you attached). Do you want to add one?

mathisono, 2024-09-23:
With out listing the changes in something like a change log! Users can make mistakes!

aanon4, 2024-09-23:
I'll think about how to log the changes. Cant display them to the user because they'd be a bunch of meaningless unified diffs.

aanon4, 2024-09-23:
If tomorrow you can verify you can no longer save tunnels with a weight of .5 that would be great.

mathisono, 2024-09-23:
I can confirm that im unable to save a tunnel with a weight of .5.

closed 2024-09-23 by aanon4