aredn/aredn #1572: Hopping Between Nodes with New UI and Old UI

K5RA, 2024-09-25:
System has   hAP ac lite (hAP)   host node with DtD link to   NanoLocoM2 (NLM2)  .   HAP is running N 20240924, NLM2 running 3.24.6.0.  HAP LAN has an R Pi 5B with latest Raspberry Pi OS and Chromium.  

If hAP is running the New UI, attempting to hop by clicking  NLM2  in Local Nodes list fails (“Not found ….”) because the browse addr is for the URL of a node with the New UI (http://<nodename>.local.mesh:8080/a/status)  searching for a node which does not have New UI.  Changing the  /a/status  to   /cgi-bin/status  in the URL allows the browse to work.  

Is this what is supposed to happen?  Appears this will not be a problem when all nodes are running New UI, but until then…?

aanon4, 2024-09-25:
Please attached the support data from both nodes - thanks.

aanon4, 2024-09-25:
Worth noting that if you look at the links themselves (hover over them in the browser) they will neither contain port 8080 or /a/status as part of the URL.

K5RA, 2024-09-25:
[supportdata-K5RA-PKR7A-202409250628.tar.gz](https://github.com/user-attachments/files/17130119/supportdata-K5RA-PKR7A-202409250628.tar.gz)
[supportdata-K5RA-HAP-FB-2024-9-25-6-27.tar.gz](https://github.com/user-attachments/files/17130122/supportdata-K5RA-HAP-FB-2024-9-25-6-27.tar.gz)

In the new UI, when I hover over the node name in the Local Nodes list, I get a box with   Link status: excellent   rather than an IP. 
I was mistaken about the IP to where the New UI tries to browse.  There is no 8080 between local.mesh and /a/status.  The 8080 is present in Old UI IPs.

aanon4, 2024-09-25:
Thanks.
If I navigate to K5RA-HAP-FB and click on the Local Node link to K5RA-PKR7A it correctly navigates me to the node running the old UI.
So ... a question and a thing to try:
Question: did K5RA-PKR7A every run a newer version of the software than it does now?
Try:  could you empty your browser cache and retry this?

K5RA, 2024-09-25:
You are correct, sir!  I was investigating what seems to be an anomaly with WiFi Scan, and reverted PKR7A from N20240920 ( I think it was) back to 3.24.6.0.  
I deleted all history in the Chromium browser, and the hop to PKR7A works fine, both with New and Old UI.
Thank you for all you do for AREDN.

closed 2024-09-25 by aanon4