aredn/aredn #1574: Local Nodes List Issue?

K5RA, 2024-09-25:
I have a hAP ac2 (HAP).  I configured Ethernet P4 and P5 as DTD, and attached one NanoStation M2 (NSM2) to each port through a PoE brick.  The HAP has firmware N20240925, and the NSM2s have 3.24.6.0.  The HAP LAN has an R Pi 5B with latest bookworm and Chromium browser.  Browsing history cleared. 

The NSM2 on HAP P4 was attached first, and it shows as a Local Node in the HAP Node Status window.  The NSM2 on P5 shows in Neighborhood Nodes list along with a  hAP ac lite  connected by a WG tunnel.  In the HAP Mesh Status display, I see the HAP with ETX of 0 and the two NSM2s with ETX of 0.1 in the Excellent block.  Using the Old UI Mesh Status of the HAP, I see both NSM2s as DTD.  

Based on the documentation, I believe both NSM2s should show as Local Nodes.  Is this an artifact of not having a recent nightly in the NSM2s?

K5RA, 2024-09-25:
Here is the Support Data for the hAP ac2
[supportdata-K5RA-HAP2-FB-2024-9-25-14-13.tar.gz](https://github.com/user-attachments/files/17137861/supportdata-K5RA-HAP2-FB-2024-9-25-14-13.tar.gz)
.

aanon4, 2024-09-25:
Please attach all the support data for the nodes in question.
Also, may I recommend you check the lat/lon for each node and make sure they're all at the same location..

K5RA, 2024-09-25:
Yes, it was a lat/lon issue.  I remember now the discussion about 100 ft or 100 meter Ethernet max cable length.  The address in the HAP2 was correct.  PKR20 had no lat/long, and PKR10 had lat/lon that was 2 miles away.  Put correct location in all three nodes, and eventually Local Node list and Neighborhood list of the HAP2 was what I expected.

closed 2024-09-25 by K5RA