aredn/aredn #1576: Meshchat download does not work

KQ6EO, 2024-09-26:
### Describe the bug
Nightly upgrade (20240925-e8fb21d )removed Meshchat from the node.  The Meshchat api was still there.  Removed api and tried to reload but received the error message shown in the screenshot.

[image]

aanon4, 2024-09-26:
How did you get "meshchat" to show up in the Downloads selector?

edkraus, 2024-09-27:
FWIW, I have one node with 20240906-f5ef0ec on it.  It was a bear to get the service packages to work, they did not auto install on the new UI.  I ended up deleting them and re-installing them after the new UI was working.  I however only install packages from my local machine, I never ask a node to install direct from some repository.  I keep all the latest firmware and packages I need in folders on my hard drive.  I don't know if your error is because trying to skip a step (download to local first) or if the newer nightly created the issue.    Ed

aanon4, 2024-10-04:
Closing as there has been no follow up.

closed 2024-10-04 by aanon4

edkraus, 2024-10-04:
the issue was with an early version of new UI, current release 3.24.10.0 did an automatic re-install of packages correctly.
Ed