aredn/aredn #1584: Local Nodes Lists Anomaly

K5RA, 2024-09-28:
I have a test bed with three UBNT "tiny" nodes linked by both rf (with high SNRs) and DtD & WG Tunnels through hAP ac nodes. 
All nodes (rf and APs) running N20240927 firmware.    Chromium browser used with R-Pi 5 to see windows. 

In the Admin window of rf node PKR7A, rf nodes PKR10 and PKR20 appear in Neighborhood Nodes list,
    with PKR10 gray and lined out. 
    
In the Mesh Status display of PKR7A, PKR10 has ETX of 1 and PKR20 has ETX of 1.1.
PKR10 and PKR20 are DtD'd the same hAP ac2 (P4 and P5 configured as DtD). 
PKR10 and PKR20 are both easily accessed from the Admin window of PKR7A. 
Why does PKR10 have a hover window that says "Link status: blocked as duplicate."

Orv, 2024-09-29:
Tim, if there are both DtD and Rf links between two nodes, the RF link will be disabled, as the DtD link will always be the better path.

K5RA, 2024-09-29:
Orv - 
I understand that DtD will predominate.  PKR7A sees both by RF and both by DtD over the same path (DtD to HAP-FB, tunnel to HAP-WH, tunnel to HAP2, then both PKR10 and PKR20 are DtD to HAP2).  Why on the PKR7A Local Nodes list does PKR10 show blocked and PKR20 does not?  And it is not really blocked at all: I can click on the blocked PKR20 entry and hop to that node.

Orv, 2024-09-29:
Remember, only that specific link is blocked; if there's an alternate route to that node, clicking on its nodename will take that route.

closed 2024-09-30 by K5RA