aredn/aredn #1585: New MikroTik RouterBOARD 952Ui-5ac2nD 5gh radio not on untill switching to 2.4

KD4WLE, 2024-09-30:
### Describe the bug
New flash - 5Ghz radio isnt 'active' until switching radios to 2.4 and back. no frequency displayed, cant hear other radios next to it. 

- [Yes ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.
Radio should be active

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
[supportdata-N4TDX-Edgewater-GW-2024-9-30-1-24.tar.gz](https://github.com/user-attachments/files/17182676/supportdata-N4TDX-Edgewater-GW-2024-9-30-1-24.tar.gz)

aanon4, 2024-09-30:
The odd thing here is that the radio should come up on 2.4 not 5 (it picks wlan0 by default, which is the 2.4 radio). Just want to check that the support data is from the device *before* you fixed the issue by switching radios on and off?

aanon4, 2024-09-30:
As there seem to be a bunch of wireguard tunnels in the support data, I'm guessing the support data is not from when the failure was observed :-(

aanon4, 2024-09-30:
The data you sent over is very odd. There's no wifi device defined at all for some reason, and there are multiple entries on the setup file which really cant be there given the code which generates them. And channel 131 isnt the default for any configuration.

If you could reflash one of your extra devices, send over the support data and a screenshot of the main page before configuring anything, that might help.

aanon4, 2024-09-30:
Just want to confirm that you were using nightly as both the initial ramdisk boot image and the initial install image, and not going into the old UI at any point?

KD4WLE, 2024-09-30:
I may have used an older ramdisk in the initial load with a newer nightly sysupgrade bin.

Yes, looking at the timestamp on the rb.elf file, indeed reads 1/12/24 and aredn-20240927-179ce13-ath79-mikrotik is in my download directory for the sysupgrade. I’ll update the rb.elf and reflash, see if the problem persists tonight.

KD4WLE, 2024-09-30:
Also:

https://downloads.arednmesh.org/snapshots/targets/ath79/mikrotik/aredn-20240928-9d091e0-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-initramfs-kernel.bin

File not found.

aanon4, 2024-09-30:
File not found - yes, there's a new nightly now with a different version number. We dont keep the old ones.

KD4WLE, 2024-09-30:
Re-flashed with today's nightly

[image]

[supportdata-KD4WLE-Setup1-2024-3-22-22-17.tar.gz](https://github.com/user-attachments/files/17195405/supportdata-KD4WLE-Setup1-2024-3-22-22-17.tar.gz)

aanon4, 2024-09-30:
Thanks for this.
This should fix the issue https://github.com/aredn/aredn/pull/1588
However .. it's a workaround as I still dont undetstand what's really at the root of this. However, I plan to do some other work on the initial setup for the next release, so I'll revisit.

closed 2024-10-14 by aanon4