aredn/aredn #1586: Some sites show trunkated IP addresses LAN DHCP screen

kp4djt, 2024-09-30:
### Describe the bug
While checking and changing some address data regarding a service I noticed that the last digit was not visible in the address area
All of the addresses on this screen showed the same issue. Appears to happen when the last 3 quads are 3 digits long. If there is a 1 or 2 digit quad (not counting the first quad) then the field reads well

- [ ] Does this issue still occur on the latest version? (**Required**)
-Yes it does, just updated to 20240928-9d091e0

### Expected behavior
All fields show show full IP, there should be no under runs of the IP fields 

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
[image]

If applicable, add screenshots to help explain your problem.
Here is a screen shot of another node with shorter fields the addresses all fit
[image]

### Additional context
This appears to be related to Chrome, and Chrome based browsers, Firefox displayed correctly Vivalde had the same issue. I did not have other browsers to test with. (Opera kept on crashing)

kp4djt, 2024-09-30:
Note screens were taken from same machine and same browser just different tabs.

kp4djt, 2024-09-30:
> There was a Issue on the "short" fields in several of the windows in the
> New UI. I think it was about 5 weeks ago, but cannot find it. It may be an
> artifact of using the High Contrast theme. Use Light theme to see all of
> it. It is also possible to move the displayed value with direction arrow
> keys to see all the characters.

I think it maybe an artifact of chrome based browsers, FF did not seem to
have issues with it Vivaldi did
and it is based on Chrome. I tried Opera but it kept on abending on me so I
trashed it and moved on.

I tried the different theme options but none of that cleared it up.

K5RA, 2024-09-30:
Light and Color Blind  themes  using R-Pi with Chromium are OK for me,  High Contrast theme is not.

kp4djt, 2024-09-30:
I do not use the high contrast either. I guess it is probably good on the
Chromebook in the van. But normal
desktop ops the regular light and dark are just fine.

K5RA, 2024-09-30:
Sorry. I was not clear. I use High Contrast all the time.   The other themes do not have the issue of IPv4s over-filling their cells because their fonts are smaller by a few points.  If I were going to spend a lot of time looking at the IPs, I would switch to Light theme.  For a one-time look, I use the arrow key workaround.

kp4djt, 2024-09-30:
Ahh, OK. It is a field space issue, I see that Tim has already pushed out a
new nightly will try it.

K5RA, 2024-09-30:
See closed issues 1363/1366 and 1376, and open issue 1389.

closed 2024-10-04 by aanon4