aredn/aredn #1596: Support for Aruba AP-105

RAndrewThomas, 2024-10-03:
### What is the hardware/platform

Aruba AP-105
PoE AP based on AR7161 CPU and AR9223 + AR9220 radios
2.4 GHz + 5 GHz
16 MB Flash, 128 MB RAM

### Links

OpenWRT ToH page:
https://openwrt.org/toh/aruba/ap-105

Supported in OpenWrt 23.05.5

### Additional context

I acquired 60 ea of these APs for my local radio club K5SRC for our area mesh project. (I got a good deal on ebay at < $3 each).

I have some experience with OpenWRT myself, having used it for a few embedded Linux experiments.  I don't intend to ask anyone else here for their effort, other than to merge my changes if/when they come to fruition.  I won't create a fork at github until I have made some progress offline.

However, if anyone here DOES want one or two of these APs I will be glad to provide them at shipping cost.

-N5RGN

kp4djt, 2024-10-04:
Andrew can you email me kp4djt@gmail.com? Trying to find folks in Stillwater to extend the mesh that way from OKC or Enid.

RAndrewThomas, 2024-10-04:
Status update:

I've added TARGET_DEVICE_ath79_generic_DEVICE_aruba_ap-105 to the build configs and gotten firmware built.  I'll be testing this weekend.

Side-note: Also in valuable conversation with kb4djt on getting our club up to speed.  Thanks!

n5rgn, 2024-11-02:
Created a PULL request "add support for Aruba AP-105"

aanon4, 2024-12-04:
https://github.com/aredn/aredn/pull/1668

aanon4, 2024-12-04:
The pull request was closed when no feedback was received to questions. Closing this too.

closed 2024-12-04 by aanon4