aredn/aredn #1601: admin ui web interface

dod1450, 2024-10-04:
### Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

dod1450, 2024-10-04:
I am no longer seeing a web GUI html page When I press the new UI I don't think the developer are seeing this?

aanon4, 2024-10-04:
Sorry - you need to provide a much more detailed description about what's going on.

closed 2024-10-08 by aanon4