aredn/aredn #1604: Typo on New UI Interface

caseydiers, 2024-10-05:
### Describe the bug

The section of the new interface where you input antenna details, clarify the antenna elevation is intended to mean the vertical tilt of the antenna.  Perhaps rename this to "vertical elevation" or add that "0° = parallel to the ground".  Having that field right after height, made me first think it was asking what the elevation was MSL (whereas height was AGL).

I think it would also be helpful to have the elevation field immediately after azimuth, as we usually do az then el.  Having height in between them adds to the confusion.

The units clarifies it a bit, but I think this could me more clear to someone that isn't used to directional antennas.

- [X] Does this issue still occur on the latest version? (**Required**)
    -  [X] YES

### Expected behavior
Someone to send me cookies to my home.  Or perhaps a cake.

### Screenshots

[image]

### Additional context

https://github.com/aredn/aredn/blob/1505d61871e57bfb09f1fcd02b193467bf8b2797/files/app/main/status/e/radio-and-antenna.ut#L459

caseydiers, 2024-10-05:
I see from the code that there may be a tool tip to clarify this, but I cannot get that tip to appear for the life of me.  Manjaro linux with latest firefox (131.0).

aanon4, 2024-10-06:
Please push the Help button in the top right corner of the dialog. It would be useful to get feedback on this clarifying inline help also. Thanks.

K5RA, 2024-10-06:
The Help text for Elevation has a typo (degress vs degrees).   

Perhaps would clarify to change Elevation Help text to read 
"Elevation angle of the antenna, in degrees above or below (less than zero) the horizontal."

Does database know about antenna beam look-down angles?  Is it correct that "elevation angle of antenna" is the orientation of the physical antenna structure?

aanon4, 2024-10-06:
The database knows nothing about electronic downtilt and I think it would be good to clarify this in the help. This is what I've got - feedback?

`Elevation of the antenna, measured in degrees, above or below (less than zero) the horizontal. Some antennas include electronic downtilt. Any such offset should be included in the number entered here.`

K5RA, 2024-10-06:
I like it!

aanon4, 2024-10-06:
https://github.com/aredn/aredn/pull/1605

closed 2024-10-14 by aanon4