aredn/aredn #1607: New UI: Unable to enter a hyphen char in text part of Local Services URL

n2mh, 2024-10-07:
### Describe the bug
A clear and concise description of what the bug is.

When entering the text part of a url in the Local Services screen, apparently the built-in templates do not allow entry of a hyphen character `-`. Thus, the full display of a MeshPhone telephone number is not allowed  - `xxx-1234`. This has been observed in Generic URL, Simple Text and Generic HTTP. I had to revert back to the Old UI in order to accomplish my task.

- [yes] Does this issue still occur on the latest version? (**Required**)

Issue observed in Production Release 3.24.10.0

### Expected behavior
A clear and concise description of what you expected to happen.

I expect the hyphen character to be allowed in order to properly display a MeshPhone telephone number. See Local Services on N2MH-Supernode-NJ for desired format.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

n2mh, 2024-10-07:
Checking further, you may have to see my example using the Mesh Status page using the Old UI. It doesn't show up in the new UI.

aanon4, 2024-10-07:
https://github.com/aredn/aredn/pull/1608

closed 2024-10-14 by aanon4

n2mh, 2024-10-17:
I manually edited the affected file, `files/app/main/status/e/local-services.ut` with the changes (3) on an operating node (hAP2) and it worked as advertised. I also did the same change on an x86/QEMU node and it worked Ok also. I also did the same treatment for the colon char `:` on both nodes and it worked Ok as well. Now, both chars are accepted when describing Local Services.

aanon4, 2024-10-17:
You didn't mention : in the original bug. I will make that valid also.
https://github.com/aredn/aredn/pull/1638