aredn/aredn #1620: PowerBeam-M2-400 Has Different Radio Model Name

K5RA, 2024-10-12:
I upgraded a two-year old UBNT PowerBeam-M2-400 from 3.22 to 3.24.10.0.  The Firmware window shows  (ath79/generic) UBNT (powerbeam-m2-xw) but the Radio block on Admin Page says Radio Model is Ubiquity NanoBeam M2 International.  

I expect Radio Model to be same as vendor M/N: PowerBeam-M2-400.

Screenshot and Support Data provided.

Hosting system is hAP ac lite running 3.24.10.0 and an RP5B running recent bookworm and Chromium.  The PBE-M2-400 is plugged in at Port 5 (DtD/PoE).  

[image]
[supportdata-K5RA-PBE400-1-2024-3-23-0-29.tar.gz](https://github.com/user-attachments/files/17348689/supportdata-K5RA-PBE400-1-2024-3-23-0-29.tar.gz)

RAndrewThomas, 2024-10-13:
This might be an upstream (OpenWRT) issue.

Your 'data.txt' and '/tmp/sysinfo/boardid' both say boardid is 0xe2c2

aredn/openwrt/staging_dir/target-mips_24kc_musl/root-ath79/usr/share/libiwinfo/devices.txt says 'NanoBeam M2 Intl" for "Device Name" when "subsystem device id" == 0xe2c2.  There is no entry in that file with "Device Name" == "PowerBeam M2*"

I don't have a PowerBeam M2 (PBE-M2-400) or I'd double check its board_id.  Interestingly that file TXT doesn't have an entry for my NanoBeam M5 (NBE-M5-400) so I will check it's board_id

K5RA, 2024-10-13:
Thank you for this information.  I have the device upgraded to 3.24.10.0, and it appears to work.  There are three other AREDN devices running on my bench, and it sees all of them.  

The box that holds the radio/antenna feed part is labeled PBE-M2-400-US.  The text below the SKU code is (K) 74ACB9FE8B4F.  The test date on the UBNT tag  is 08/23/20.  I bought the node from a reseller about 02/25/2022.  The nameplate on the back of the feed assemble has the following markings: 
M/N:  PBE-400-M2
FCC ID: SWX-NBM2HP
IC: 6454A-NBM2HP
Could UBNT have used a NanoBeam board in the PowerBeam product, and listed the old FCC and IC certifications?

RAndrewThomas, 2024-10-13:
Oh, I bet you're absolutely spot-on there.  I just looked on both a PowerBeam M5 (PBE-M5-400) and a NanoBeam M5 (NBE-M5-400) and they BOTH have the FCC ID: SWX-NBM5HP.  Now I've got two board_id's to check ;)

K5RA, 2024-11-03:
I guess it is what it is.  I know what radio I have, and will put that in the description.  No need to leave this issue hanging out.

closed 2024-11-03 by K5RA