aredn/aredn #1621: Deleting Tunnel Entries in New UI Causes issues

k1ky, 2024-10-12:
### Describe the bug
A clear and concise description of what the bug is.
Deletion of Tunnel Server Entries can cause a ripple effect between the New and Old UI where the Network assignments don't synchronize and can cause issues with tunnel clients down the line after the deletion.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes N-20241011

### Expected behavior
A clear and concise description of what you expected to happen.
Deletion of tunnel server entries should probably be handled by merely blanking the description fields or putting a placeholder so the Network IP addresses don't change. Any entry from the deleted and further down the line causes a very disruptive "re-ordering" of the assignments, which would cause the end users to change their Tunnel Client Network entries in order to remain connected.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
I chased this around between the Old and the New UI until I became crosseyed!  Definitely DO NOT save any changes made in the Old UI!

aanon4, 2024-10-12:
Yes. Dont use the new UI and then go and use the old UI. It wont go well and I believe we advised against this in the release notes. This problem will go away because the old UI is going away.

k1ky, 2024-10-12:
I understand that all too well now, but the issue of the "reassignment" of the remaining tunnel entries is also a problem as described above with the Tunnel Network addys changing.  I don't think that is an Old vs new UI issue?

aanon4, 2024-10-12:
Not sure I understand then. If you delete a tunnel entry then address will absolutely be reused.  The other end of the tunnel should also be deleted. Assuming this is done, is there still an issue?

k1ky, 2024-10-12:
From what I observed, the "remaining" Tunnel Clients that are in the Tunnel Server section Network IP addys are reassigned to "backfill" or re-use the deleted network addy.  In some cases it appeared that some entries may have even changed the order of their entry, but that may have been a result of flipping back and forth.

aanon4, 2024-10-12:
The new UI has no Tunnel Client or Tunnel Server section. If you are referring to the old UI you *cannot* use it if you do anything with tunnels in the new UI. It will not even show you the correct information anymore.

k1ky, 2024-10-12:
Yes, as I learned that the hard way! I "assumed" that they were pulling from the same data file or something like that and would reflect the same information. Lesson learned! Will be interesting to see how to prevent this especially with such a mixture of Old vs New Firmware in place throughout our networks.

closed 2024-10-12 by aanon4