aredn/aredn #1627: Wifi Signal does not display both LOCAL and REMOTE bars

KenHorse, 2024-10-13:
When using WiFi Signal, there is no display for the REMOTE side. Also notice LOCAL display moves up and down (see screenshots)
[image]
[image]

1013 Nightly build. Seeing issue on Powerbeams M5 XW, 5AC 400 and 5AC Gen2. Browser is latest FireFox

aanon4, 2024-10-13:
Please provide support data for both end of the link you are testing. Grab this data immediately after running the test. Thanks.

KenHorse, 2024-10-13:
[supportdata-5AC Gen2-2024-10-13-20-43.tar.gz](https://github.com/user-attachments/files/17357028/supportdata-5AC.Gen2-2024-10-13-20-43.tar.gz)
[supportdata-M5 XW-2024-10-13-13-42.tar.gz](https://github.com/user-attachments/files/17357031/supportdata-M5.XW-2024-10-13-13-42.tar.gz)

aanon4, 2024-10-13:
Thanks. Which way were you running the test just before you got the support data?

aanon4, 2024-10-13:
Also, could you redo the screenshots but without cropping out the rest of the UI? Thanks.

KenHorse, 2024-10-13:
I re-ran the tests from both ends again and grabbed supportdata again. Also screenshots from each end
[image]
[image]
[supportdata-5AC-GEN2-2024-10-13-21-24.tar.gz](https://github.com/user-attachments/files/17357114/supportdata-5AC-GEN2-2024-10-13-21-24.tar.gz)
[supportdata-M5-XW-2024-10-13-14-23.tar.gz](https://github.com/user-attachments/files/17357115/supportdata-M5-XW-2024-10-13-14-23.tar.gz)

aanon4, 2024-10-13:
You need to select a node - there is no remote average. Did you try that?

KenHorse, 2024-10-13:
Dang, didn't even occur to me...duh.... Yea, that works fine. Close this!

aanon4, 2024-10-13:
I will update the Help.

aanon4, 2024-10-14:
https://github.com/aredn/aredn/pull/1628

closed 2024-10-14 by aanon4