aredn/aredn #1633: Supernode Neighborhood Invasion

K5RA, 2024-10-16:
What are node setup parameters in 3.24.10.0 to prevent what appears to be an entire Supernode neighborhood from being linked to our local mesh island?  
Our island mesh has about 40 nodes and 100 devices.   When I activate my WG Tunnel client served to me by a gateway to a nearby Supernode, our mesh node count exceeds 1100 and the device count exceeds 3200.  I operated with this link active from 26 Sep until about 14 Oct without any difficulties.  Our mesh island was serious impaired until I deactivated the link. 

My system is hAP ac lite with 3.24.10.0 and R-Pi 5B with up to date bookworm and Chromium.  I serve out four active WG tunnels and am client to three others.  The Supernode gateway is built on QEMU X86 platform.

aanon4, 2024-10-16:
https://docs.arednmesh.org/en/latest/arednHow-toGuides/supernodes.html

K5RA, 2024-10-16:
Tim - Thanks for the link.  I do not have a Supernode.  Our mesh island has had a link to one SN for months with no problems.  However when I connect one of my nodes to a second SN, the counts go crazy.  I will work with the SN owner.

aanon4, 2024-10-16:
You never connect a non-supernode to a supernode using a tunnel (see the doc link I referenced earlier). I assume this is your problem.

K5RA, 2024-10-22:
Yesterday I deleted then reconfigured the WG Tunnel Client served to me by a local Supernode's gateway node.  After rebooting my node, the tunnel came up, the gateway node appeared in my Neighborhood Nodes list,  AND our mesh island's nodes and devices counts increased only slightly.  

I am still not able to hop successfully to any Supernode in the Neighborhood Nodes list except the two that are already linked to our mesh island.  Appears to be a DNS issue.  Message appears immediately after clicking a node name:  "This site can't be reached.  Check of there is a typo.... DNS_PROBE_FINISHED_NXDOMAIN" .  

Probably should close #1633 and open a new one for this if cannot solve it here.

K5RA, 2024-10-23:
Tonight I discovered that the Cloud Mesh Internal Services button was disabled.  Switched it to green, rebooted the node (required to make it work, but does not happen automatically), and I can see Supernodes all over the map.

K5RA, 2024-10-23:
We do not know how we got the entire node/device table from some Supernode on our mesh, but until it happens again this issue is closed.