aredn/aredn #1634: cgi-bin/api was not unused

r1de, 2024-10-16:
Thanks for breaking my map.
How the heck do I get the network topology info now?

aanon4, 2024-10-16:
/cgi-bin/sysinfo.json is your friend. It's much simpler and pre-dates the api endpoint. With various options it should get you want you want. Let me know if it doesn't.

r1de, 2024-10-16:
sysinfo.json cannot give me a list of all the _node_ IP's on the network, it only tells me info about _that_ device and how _it_ is linked to it's direct neighbors.
(I just went and looked at the code for sysinfo.json again, and your right it hasn't changed much in years.)

cgi-bin/api did, and it could give just a list of the nodes, no hosts or phones or other devices, just the _routers_.
It also had how many hops away a node was from the "localnode" and a lot of other info about the state of the network and OLSR.

What that file taking up too much space or something? Why?

aanon4, 2024-10-16:
Does http://localnode.local.mesh/cgi-bin/sysinfo.json?hosts=1 give you want you want?

r1de, 2024-10-16:
Not even close, that's why I used the API.
It was like window into OLSR, especially for RF links.
Why did it have to go away?

r1de, 2024-10-16:
That makes sense.
I'll go try and pick apart the old file and see which parts I needed.
Thx Tim.

r1de, 2024-10-16:
Actually it should be quite simple.
When I would call `api?mesh=topology`, all the api was doing was returning the json from:
`http://127.0.0.1:9090/topology`.

This part specifically:
```
elseif comp=="topology" then
  info['pages'][page][comp]=fetch_json("http://127.0.0.1:9090/topology")
```

Output was like this (which doesn't matter all that much):
[image]

So all that is really needed is to just pass-thru that data via something like `sysinfo.json?topology=1`

aanon4, 2024-10-17:
Let me know if there's anything else missing that you need.

r1de, 2024-10-17:
I'll update after it builds and I install it tomorrow
I'll need to only change a few lines in my code.
I think that should be it tho.
It is the ancient port 9090 problem again, almost. :smile: 

Thx Tim.

r1de, 2024-10-17:
Works as before, just in a new place.
Now to clean up my quick and dirty test edit in _my_ code.
Thanks Tim!
I'll let you close this issue.

closed 2024-10-24 by aanon4