aredn/aredn #1636: Add UI page(s) for log files

Orv, 2024-10-17:
**Is your feature request related to a problem? Please describe.**
While the ability to send logs to a log server is valuable, many AREDN users either don't have enough nodes to make it worthwhile, or lack the skill set to neither set up a log server, nor ssh into a node.   This class of users (the majority) could use a simpler way to gain access to the log files for troubleshooting

**Describe the solution you'd like**
A UI-based log page, either one page per type of log, or a log page with a separate stanza for each type of log

**Additional context**
Maybe flag errors as red and warnings in yellow, to make identifying problems easier.
Given that this class of user is most probably not a systems or network expert, a help entry describing what might cause each error could be useful, too.   Alternatively point them to the documentation system on arednmesh.org for information on logged issues.

aanon4, 2024-12-04:
https://github.com/aredn/aredn/pull/1744
See comments in the PR about feedback on enhancing this.

aanon4, 2024-12-04:
Going to close this as completed. Open a new one when we have a plan to make this more fancy.

closed 2024-12-04 by aanon4