aredn/aredn #1640: GL-iNet GL-E750 Unable to Flash into Aredn

aa3jc, 2024-10-21:
Model GL-E750c4 running factory firmware version 4.0, Firmware Type 0312release4, compile time 2024-04-10 18:47:38(UTC+8:00).

Aredn's website indicates there is a "Factory" build and "Sysupgrade" build.  When you click on "Factory" build, page refreshes to a "404 Not Found" page.  If you click on "All Downloads" and search for "ath79/nand target, I can find "Sysupgrade" builds but no "Factory" builds. This is true for previous versions for the GL-iNet GL-E750.
Attempting to flash the GL-E750 with just the "Sysupgrade" firmware leads to failure as follows:

From Web Interface, click "SYSTEM", "UPGRADE", and "LOCAL UPGRADE", Then from Windows Explorer " dragged and dropped aredn-[3.24.10.0](http://3.24.10.0/)-ath79-nand-glinet_gl-e750-squashfs-sysupgrade.bin into the file upgrade box.  After it verified file new window appears,  with a yellow box with the warning "This is not official released firmware".
Then below that is a box with "Firmware Verification"
Version:   Unknown
SHA256.     "With a long string of numbers & characters ".
Verification Results.  "PASS"
Keep Settings. "Unchecked"
Then a blue box "Install"  
Once you click "Install" computer screen clears, and Tells you it is Downloading, then changes to Upgrading.  Also, you can close that window.  
Display on the GL-E750 changes to " Upgrading Firmware" and stays on this for over an hour with no visible indication it has rebooted even once.
After an hour I manually shut down and restarted it, but the display gets stuck partway through "booting".  Let it run for over another hour and the sceen still indicates "booting".  I tried using the U-Boot method of flashing to Aredn, and got the same failure if flashing results.  
I can restore the GL-iNet factory Firmware, so I attempted to flash it to Aredn a third time but let it run for 3 hours with the same results.
Over a year or more ago I attempted to flash it to Aredn (I don't remember the version of Aredn at that time) but at that time Aredn had a factory build and the system upgrade build. 
At that time, the unit rebooted into the first Aredn build and I went into the node to flash the system build, in large red letters Aredn Claimed that it was an UNSUPPORTED DEVICE.  I believe I opened a Ticket on this issue at that time with no response back from Aredn Support. Point is I was able to get to that point in the process using the "Factory" build.

This GL-E750 was purchased over a year ago and there is nothing indicating a version 1 or version 2.

Now, the GL-E750 never seemed to do the normal 2 reboots.  After an hour tried setting computer to DHCP but never got a ping response from either [192.168.1.1](http://192.168.1.1/) or [192.168.8.1](http://192.168.8.1/)  with or without a static ip of [192.168.1.10](http://192.168.1.10/) or [192.168.8.1](http://192.168.8.1/) and with no ping response at any time.  Able to shut down and boot into u-boot and tried the Aredn Firmware from that method but again it get's stuck on "system upgrade" and never rebooted.  At that point manually shut down the node after 30 minutes, powered back on but again could not ping or get the web interface up.
Able to enter u-boot and reflash the latest GL-iNet factory firmware without any problem.
NO LOG AVALIABLE.

aanon4, 2024-12-04:
I would need the hardware to debug this.

KD9PCD, 2025-04-10:
Is there anymore updates with this? I’ve got the hardware and would be willing to send it out if needed.

aa3jc, 2025-04-11:
No,  I attempted to again for the third time flashing the one I had purchased OVER 1.5 years ago, and it still fails on flashing to Aredn.  After trying for an hour, I ended up flashing it back to GL-INet factory firmware and it works perfectly as it was intended!
Trouble Tickets were opened but I have not heard back on those tickets.
Aredn used to require 2 firmware's to flash new devices, the Factory and System firmware, but the last few production versions have gone to a single file flash, since the factory stock firmware is already running a highly modified version of WRT.
If you do try it and succeed let us know!

Joe Caltabiano-AA3JC
PEMA-ACS Eastern Area Section Chief
________________________________________________
Joe Caltabiano AA3JC ComL, ComT, Auxcom, ACS
Eastern Area PEMA-ACS Section Chief
215-237-7991 (cell)
MCAR Mesh IP Phone: 215-3103
PEMA-ACS Mesh IP Phone: 215-2001
HOIP: 100294 (cell)
H.H.: 5500000968 (cell)
***@***.***
***@***.***
***@***.***

KD9PCD, 2025-04-11:
Well, I was able to load AREDN onto the device. The OLED screen on the device doesn't match anything going on within the device with the AREDN firmware, but after I loaded it I switched my computer back to a DHCP IP and reached the node on 192.168.1.1. I could not reach the node on localnode.local.mesh. I also have no radio function and it will not assign an IP address outside of 192.168.1.1, even after unplugging and plugging back in. I tried multiple computers. The three main issues I'm having with it now are the IP issue, the radio issue, and the OLED screen just stuck on saying "BOOTING." 

I really don't need much info on the OLED screen, even just the battery level would be fine. I am able to access the system log and the full tools menu, too. I installed the latest stable version, 3.25.2.0.

aa3jc, 2025-04-11:
Exactly the same as when I flashed it last.  I even tried a nightly with identical results.

If your a member of Aredn on SLACK you may want to post your findings there as it seems most of the Aredn support hang out there.  They probably will tell you " to open a ticket on GitHub.

The more people who report it the better.....

Joe Caltabiano-AA3JC
PEMA-ACS Eastern Area Section Chief
________________________________________________
Joe Caltabiano AA3JC ComL, ComT, Auxcom, ACS
Eastern Area PEMA-ACS Section Chief
215-237-7991 (cell)
MCAR Mesh IP Phone: 215-3103
PEMA-ACS Mesh IP Phone: 215-2001
HOIP: 100294 (cell)
H.H.: 5500000968 (cell)
***@***.***
***@***.***
***@***.***