aredn/aredn #1641: Enhancement: Tool to View GPS Data

K5RA, 2024-10-21:
Propose an enhancement to AREDN for a tool to be used with data from a GPS dongle to provide a display like that of the   XGPS app  in a Raspberry Pi.  Tool would provide list of satellites available and indication if used or not, graphic showing GPS constellation, time and location values and error estimates, etc.  See screen shot.  
[image]

K5RA, 2024-10-21:
Tool would be used with nodes like hAP ac lite and hAP ac2 that have USB-A port to accept the GPS dongle.

aanon4, 2024-10-24:
I'm inclined to leave this to someone to write an external package for rather than including it by default.

aanon4, 2024-12-04:
Someone should write a package for this feature.

closed 2024-12-04 by aanon4