aredn/aredn #1642: NAT subnet bigger than /24 breaks UI

rjacoby00, 2024-10-22:
### Describe the bug
On the new UI, attempting to set the LAN subnet bigger than /24 will produce the following error:

```
ERROR: left-hand side expression is null

In netmaskToCIDR(), file /usr/share/ucode/aredn/network.uc, line 69, byte 19:
  called from function getDHCP (/usr/share/ucode/aredn/configuration.uc:223:55)
  called from function main (/app/main/status/e/network.ut:177:41)
  called from function call ([C])
  called from function render ([C])
  called from anonymous function (/app/root.ut:559:18)

 `        switch (m[i]) {`
  Near here --------^
```

- [x] Does this issue still occur on the latest version? -> Tested on 3.24.10.0

### Expected behavior
Firmware should allow for LAN subnets of any size.  Old UI states `LAN netmask must begin with 255.255.255` when attempting to use a larger subnet.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

[image]

aanon4, 2024-10-22:
This should be fixed when this PR goes it.
https://github.com/aredn/aredn/pull/1548

aanon4, 2024-10-29:
Please verify the new implementation.