aredn/aredn #1643: Adding DHCP option #42 (ntp-server) causes loss of WAN access and other odd behavior.

convolved-KY4G, 2024-10-23:
### Describe the bug
Adding DHCP option #42 (ntp-server) causes loss of WAN access and other odd behavior such as not being able to access http://localnode.local.mesh/a/status.  While the DHCP option is present, the node can only be accessed by typing it the IP address of the node.  As soon as the option is removed, WAN access is restored and http://localnode.local.mesh/a/status will work again.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes, this issue was observed on the latest release 3.24.10.0

### Expected behavior
I expected WAN access and to the page http://localnode.local.mesh/a/status to work as normal regardless of setting the DHCP options. I also expected the value of the tag to be passed along to a VOIP phone, but it does not appear to be functioning as intended.

### Screenshots
[image]

[image]
[supportdata-KY4G-GO-KIT-2024-10-22-21-56.tar.gz](https://github.com/user-attachments/files/17484547/supportdata-KY4G-GO-KIT-2024-10-22-21-56.tar.gz)

### Additional context
I discovered this while trying to make the node serve as an ntp-server for a Grandstream  VOIP phone.  The phone could not access the specified ntp-server no matter what I have tried. The package WhereAndWhen has been installed and the node properly obtains time information from the GPS antenna.  As far as the tag option value goes, it doesn't matter if I set it to point to the node, or if I point it to us.pool.ntp.org, the same loss of WAN access occurs.  I have tested this with or without the phone connected and I get the same result.

aanon4, 2024-10-23:
ntp-server must be a valid IP address and not a FQDN. See rfc2132 for details.
We don't do DHCP option validation yet (which is already a pending issue).

convolved-KY4G, 2024-10-23:
You are correct, the FQDN is only supported in IPv6. Replacing the FQDN with the LAN IP stopped the behavior where WAN access was lost.   http://localnode.local.mesh/a/status works normally too.  The VOIP is still not picking up the ntp-server even though it set to accept the option #42 override; however, that is best dealt with in the forums and not here.  Thanks for the quick follow up.  I am closing the ticket as DHCP option validation is handled in a previously filed issue.

closed 2024-10-23 by convolved-KY4G