aredn/aredn #1645: Add Generic 18 degree sector to antenna list

convolved-KY4G, 2024-10-24:
This is an enhancement request, not a problem.  The new UI wants me to select an antenna, but no suitable antenna in the list matches this 18 degree sector antenna that we use for AREDN deployments.

**Describe the solution you'd like**
Add Generic 18 degree sector" to antenna list, except substitute the degree symbol

**Describe alternatives you've considered**
Alternatively, leave it set to None or Unknown

**Additional context**
This is a link to the antenna. [https://www.amazon.com/Tupavco-TP511-Antenna-Wireless-Directional/dp/B008FH5UTG/ref=sr_1_6?crid=1S0SDQ1NV0323&dib=eyJ2IjoiMSJ9.OFZwo-t8VMCLPzJSSXSt8Qm_5MbXfE6LhvwGthrjvFtXqrSH-b4u4KLHbpb5p4xKoIX8ypUxEy43W29K11jZvr7Jro9WnViKODh8DKUbrus_bJlb8uSDNwEsvcJPL5neYWarW4tw5Cs9Djv6DAde80HdIy0GNDK9e3TIXZ_M_Jdo4c9VCvfzHN2L8beME-Jj76XRvQotVOWkInUiiqlGVsgDrhtcbmoRwVpUi0gxl1c.axVZw2Ph0s_jWa9EAZgAF2wBkc2m21OhxMlMp5UGlWg&dib_tag=se&keywords=tupavco+tp513+yagi+wifi+antenna+2.4ghz&qid=1729733448&sprefix=tupavco+2.4%2Caps%2C194&sr=8-6](url)

aanon4, 2024-10-24:
I would rather add this specific antenna if that's what you use. Any reason not to?

convolved-KY4G, 2024-10-24:
I am fine with your recommendation.  We have several of them and they work well. Thanks!

aanon4, 2024-10-24:
https://github.com/aredn/aredn/pull/1646

convolved-KY4G, 2024-10-25:
I tested this with the latest nightly build and it works. Thanks!

closed 2024-10-25 by convolved-KY4G