aredn/aredn #1649: Ubiquiti PBE-5AC-500 Firmware check failed on flashing

combatlord, 2024-10-25:
### Describe the bug
Try to flash new Aredn Firmware via TFTP-Client results in "Firmware check Failed"

Tried with the following FW Versions:
aredn-3.24.4.0-ath79-generic-ubnt_powerbeam-5ac-500-squashfs-factory.bin
aredn-3.24.6.0-ath79-generic-ubnt_powerbeam-5ac-500-squashfs-factory.bin
aredn-3.24.10.0-ath79-generic-ubnt_powerbeam-5ac-500-squashfs-factory.bin

Nightly:
aredn-20241024-7d9ac7e-ath79-generic-ubnt_powerbeam-5ac-500-squashfs-factory.bin

I also downgrade the AirOs to XC.v8.7.0.42152.200203.1256.bin but the results are the same

### Expected behavior
Flash successfull Aredn Firmware without failure.

### Screenshots
[image]
[image]

### Additional context
I see that other Powerbeams have the same failure, so it seems that is a generic bug.

aanon4, 2024-10-25:
Please follow these instructions: https://docs.arednmesh.org/en/latest/arednGettingStarted/installing_firmware.html#ubiquiti-802-11ac-first-install-process

K5RA, 2024-10-25:
Daunting.

combatlord, 2024-10-26:
OK, thanks for the tip. I'll try it and give a feedback.

combatlord, 2024-10-26:
It works like in @aanon4's link.
I didn't realize that this chapter was for the powerbeam.
So that pointed me in the wrong direction.

Thanks to aanon4.
It's not a bug, it's user error on my part.
Sri.

closed 2024-10-26 by combatlord