aredn/aredn #1653: Update from 10.24.10.0 to NB 20241027 resulting in first boot

k6ccc, 2024-10-27:
Upgraded 10 nodes from 10.24.10.0 to NB 20241027, and node would NOT reconnect to AREDN network, and the four that were connected via LAN were no longer reachable nor ping via the previous IP address.
So far I have checked several of the nodes that I can easily physically access and all are in first boot mode (192.168.1.1 IP and default settings) except node name retained.
Eric - KG6WXC reporting the same issue.

Node types updated:
Rocket XM M5
Rocket XM M3
Rocket XW M5
GL-inet AR750
LHG 2nd-XL
LHG 5nd
LHG 5HPnd-XL

aanon4, 2024-10-27:
Anyway you could get the contents of /etc/config and /etc/config.mesh for me to look at? I realize support data is not an option.

aanon4, 2024-10-27:
I've take the nightly build offline until we resolve this.

k6ccc, 2024-10-27:
Having spent a little time, it gets more strange.  It's not really in first boot mode.  A lot of he settings are in fact remembered - sort of.  For example on the main screen, the RF is not set, nor is the location, but in ether case, if you select that field to edit it, the settings are correct.  But back to the main screen and they are not set - and it's telling me to set the location.  Reboot does not change that.  I was able to get support data, so here it is
[supportdata-K6CCC-AR750-Portable-2024-10-27-17-58.tar.gz](https://github.com/user-attachments/files/17535008/supportdata-K6CCC-AR750-Portable-2024-10-27-17-58.tar.gz)

aanon4, 2024-10-27:
Thanks. It looks like the root of the problem is because the node doesnt have a description.

k6ccc, 2024-10-27:
Added a description.  Did not show, but trying a reboot now.

k6ccc, 2024-10-27:
Adding description and reboot did not help.
This was on a different node (easier to gt to LAN wise).  If you click on the fields to edit, the data is all there correctly, but does not show on the main page.  Here are some screen captures and a support file from this second node.
[supportdata-K6CCC-RB-LHG-5nD-2024-10-27-18-19.tar.gz](https://github.com/user-attachments/files/17535060/supportdata-K6CCC-RB-LHG-5nD-2024-10-27-18-19.tar.gz)
[image]
[image]
[image]
[image]
The screen captures were before adding the description.

aanon4, 2024-10-27:
Let just stick with one node of this will be confusing for me.
Can you send me the support data after you added the description? Thanks.

k6ccc, 2024-10-27:
I should also point out that I updated a hAPac3 from a nightly about a week or so ago to 20241027 without any problem at all.  All the ones in trouble were from the production release.

k6ccc, 2024-10-27:
No problems on sticking with the one node.  i will stick with the LHG 5nD because it's easier for me to transfer data (this computer vs the laptop).  Here is the latest support data...
[supportdata-K6CCC-RB-LHG-5nD-2024-10-27-18-33.tar.gz](https://github.com/user-attachments/files/17535074/supportdata-K6CCC-RB-LHG-5nD-2024-10-27-18-33.tar.gz)

aanon4, 2024-10-27:
The root cause. When I transferred the old "_setup" data into the new "setup" uci format, variable which were blank get stored as 'nil' (ie. not there) in the new format. There are a few places this matters, and unfortunately those places weren't blank on my test nodes. I will rummage through them all and make sure I get them all (turns out there are more than just the description).

You should be able to revert to 3.24.10.0. I dont know if you've tried this anywhere yet.

k6ccc, 2024-10-27:
Trying 3.24.10.0 on the LHG 5nD now

k6ccc, 2024-10-27:
Wife dragged me out for lunch, so be a while.  However, note that all these nodes had a description prior to the upgrade.

k6ccc, 2024-10-27:
Also, before I left, tried 3.24.10.0 on the LHG 5nD.  It reported firmware wrong.  Common on this node from Firefox, but usually works from Chrome.  Did not work.  Unable to find the "allow dangerous software" switch, so could not try that (nothing under advanced options).

ab7pa, 2024-10-27:
> Also, before I left, tried 3.24.10.0 on the LHG 5nD. It reported firmware wrong. Common on this node from Firefox, but usually works from Chrome. Did not work. Unable to find the "allow dangerous software" switch, so could not try that (nothing under advanced options).

You can always use the docs "search" feature to find it, but it's in the Node Admin guide.

[image]

aanon4, 2024-10-27:
We'll leave how two different browser could decide a node hardware is different for another day.

aanon4, 2024-10-27:
https://github.com/aredn/aredn/pull/1654

k6ccc, 2024-10-27:
Advanced options does not show on any of these nodes with either browser.
[image]

k6ccc, 2024-10-27:
OK, found it - click on the words, not the expand arrow...

Duh!

aanon4, 2024-10-27:
Its not an expand arrow. I've never seen that on any node on any browser before in fact.

k6ccc, 2024-10-27:
Nope.  Brain fart on my part.  That could be an expand on a lot of other things.

ab7pa, 2024-10-27:
It's the bottom arrow of a scrollbar -- top arrow also visible

k6ccc, 2024-10-27:
OK, on the LHG 5nD, I was successfully able to reload 3.24.10.0  Interestingly, it initially came up in first boot mode, then disappeared (I had a continuous ping to 192.168.1.1).  Told my router to renew the DHCP lease (router gets an IP from six different nodes each on different VLANs), and it came up with the previous IP (10.165.92.249).  All the settings appear to be restored (except the description), and the RF link to my hAPac3 came right up.

k6ccc, 2024-10-27:
So far, each node that I have attempted to reload to 3.24.10.0 has worked.  A few have lost a field or two, but so far, all have been recovered.  Couple more to go.  There are two I wont be able to attempt until Tuesday when I am back at work.

aanon4, 2024-10-27:
Any changes made after the update will be lost when you revert them (not usually the case, but this time)

k6ccc, 2024-10-27:
I had made any changes after the failed updates.

aanon4, 2024-10-27:
If you can try tomorrow's build, that'd be great. Tried it on my LHG 2ND (didnt have a 5 available) and it works fine.

k6ccc, 2024-10-27:
Will do.  Not gonna shotgun 10 of them at once!

aanon4, 2024-10-27:
Yeah .. please dont do that with nightlies.

r1de, 2024-10-27:
@aanon4 I spun up an AREDN x64 VM and it is happening on that also.
Going from 3.24.10 to the nightly, using either the new or old UI seems to result in a first boot condition.
Since it's a VM thought, I can easily go back to 3.24.10.
If you still need it, which support data do you need? After the upgrade or before?

aanon4, 2024-10-27:
@r1de I think we're good (PR already merged with fixes), but thanks.

k6ccc, 2024-10-29:
NB 20241028 working correctly on all 12 of my nodes that I can access.  The other two at work will get redone tomorrow morning.

Thanks for the fix.

closed 2024-10-29 by aanon4