aredn/aredn #1663: Nightly 20241030 on lhgg-5-ac-xl does not display LAN DHCP lease.

nc8q, 2024-10-31:
### Describe the bug
Nightly 20241030 on lhgg-5-ac-xl issues but does not display LAN DHCP lease.

Yes, as of 20241031 09:49 UTC.

### Expected behavior
Expected the lease to be displayed

### Screenshots
http://nc8q.ddns.net:4880/20241030

### Additional context
None, yet.

kodiakastu, 2024-10-31:
Hello, I'm having the same issue on a VM. 

It seems that the gui DHCP range is "offset" from the subnet. On my side, DHCP is working and i'm receiving addresses on the right subnet, but the gui is thinking the DHCP should be on an other one and nothing is showing on the webpage.

[image]
[image]
[image]

[supportdata-VA2LRM-LPLN-Q01-VM00-2024-10-31-9-37.tar.gz](https://github.com/user-attachments/files/17588496/supportdata-VA2LRM-LPLN-Q01-VM00-2024-10-31-9-37.tar.gz)

aanon4, 2024-10-31:
Always fun to find out the values we store were always wrong but the bug hidden by other things.
https://github.com/aredn/aredn/pull/1664

kp4djt, 2024-10-31:
I see the same thing on a hAP ac lite I use on the road. Just updated it last night, I have 4 devices on the LAN 3 wired and one WiFi, but only two IP assignments shown.

kodiakastu, 2024-11-01:
Great to see you're always that fast LOL! Do you know when this will be merged ?

aanon4, 2024-11-01:
Will be merged this evening.

kodiakastu, 2024-11-01:
Everything is back to normal, Thanks!

closed 2024-11-01 by aanon4