aredn/aredn #1667: Any plans on how to approach new WiFi hardware?

Djfe, 2024-11-02:
For crossreference:
https://www.arednmesh.org/content/any-plans-how-approach-new-wifi-hardware

I just heard about this project by chatting with someone online.
I'm working a lot with another OpenWrt fork: https://github.com/freifunk-gluon/gluon

What is your question
I'm curious:
What are your plans regarding support of new/modern hardware plattforms (targets)?
ar71xx/ath79 (and by extension ipq40xx) aren't going to stay around forever.
OpenWrt keeps on shifting towards Mediatek based targets because they actually work together with OpenWrt on hardware support while Qualcomm is rather working against them, I feel.
Mediatek/mt76 on AREDN doesn't support features like 5MHz channels, yet. afaik
Maybe you could talk to the OpenWrt/Mediatek devs involved (over openwrt devel irc; or linux-wireless or some other kernel related mailing list) to make that happen. (The way greearb made IBSS mesh support happen for ath10k-ct for the community)

Additional context

Why do you need new targets? Because old targets will eventually stop being sold/produced and fail because of age. This is a gradual process which Freifunk already fell victim to. You still have a few years to fix this, since Aredn hardware has a longer lifespan and is being sold for years longer than your typical consumer hardware people run at home.
You have to switch over to new targets eventually. So brainstorming this is essential for the project to last. Like for consumer hardware it will get harder and eventually impossible to buy new devices still using old targets. ath79 is dead for new consumer hw. ipq40xx and ramips-mt7621 will meet the same fate. Their high end successors mediatek-mt7622 and ipq806x are already fairly dead (not sold anymore since they are high end and high end switched faster).
Mediatek filogic serves low end as well as high end and is a potential candidate.

I feel like ath11k wifi: qualcommax-ipq807x (and by extension qualcommax-ipq60xx and qualcomm-ipq50xx) aren't good candidates, because Qualcomm is far from a future-proof thing for new targets. Regulatory Domain updates require firmware updates now (stupid management/legal decision, I bet. Firmware Updates have to be signed, so Qualcomm has the final say in how hardware can be used). The closed source firmware handles regdm now instead of the linux kernel (enforced by ath11k and ath12k). You can choose a country using iw, but I don't think much more than that. Firmware updates will probably be canceled by Qualcomm like they did for older targets -> that would lead to no regdm updates.
ath11k and ath12k have no open source firmware anymore. Candela Tech only developed open source firmware for ath10k (under NDA). I think the company Candela Tech might have switched to Mediatek as well (I couldn't find any source for that, this statement might be false).

OpenWrt devs like nbd that worked on ath9k in the past (opening up the atheros WiFi firmware), have been working on mt76 since the shift to ath10k or around that time.
nbd held a talk in Dec 2015: [Wireless Drivers: Freedom considered harmful?](https://www.youtube.com/watch?v=hiUosbhR0Wo)
https://fahrplan.events.ccc.de/congress/2015/Fahrplan/system/event_attachments/attachments/000/002/854/original/slides.pdf

Regards
Djfe

ae6xe, 2024-11-02:
@Djfe ,   long term availability of hardware is import to our community.  Our mode of operation has followed from OpenWrt support.  We have added extended channels in ham radio allocations with 5/10 MHz channels if not already available.  We were pointed in  the right direction by greerb and Sebastian@ddwrt for ath10k channel width implementations.   In Jun 2022 greerb communicated that he had stopped work on ath10k and had asked our interest in moving to MTK ax radios, suggesting CT direction at the time.    

We are happy to further collaborate and will take a look at the mt76 device options.  I think this is in our best interest to do in adopting ax devices.    10 MHz channel width is widely used along with the extended channels, critical to have in our community.   What are the hurdles, if any, to get 10Mhz width and  extended channels working in mt76?   

The ZYXEL AX1800 looks like a good first device to consider.    The devices we deploy are generally on a tower outdoor with a <=2 ft dish or sector.  

Joe AE6XE

n5rgn, 2024-11-03:
Zyxel AX1800 is a class or series of product.  Their model number NWA55AXE looks like their only AX1800 model that's both outdoor rated and currently supported by OpenWrt.

n5rgn, 2024-11-03:
As to open source and/or the ability to configure the radio for amateur allowances, I see the most likely options as: (a) reverse engineering or (b) paying a chip maker to write an open source driver.  Yeah, I guess we could petition or or pay them to at least add the features we need, but open source drivers are the best option.

And I think the outlook for chip maker cooperation just gets gloomier the more the BRCMs and QCAs of the world buy up smaller makers and their patent portfolios.  Broadcomm doesn't care about their customers (see VMware) as long as they maintain a big market.  Qualcomm cares mostly about the mobile market, but also doesn't see the need to cater to a niche.

Djfe, 2025-08-13:
I just ran across these 2 commits, there might be more
https://github.com/torvalds/linux/commit/996c15bd30a9caf5d3a32414a28503f3389fc96e
https://github.com/torvalds/linux/commit/2400dfe23fa91612c18c6c8d8a5b8164ff98836c

Apparently Linux dropped support for 5/10 MHz channel width for wifi with release 6.7
OpenWrt 24.10 uses mac80211 from Kernel 6.12
So the last OpenWrt release you are able to use with those channels widths might actually be OpenWrt 23.05 these days.

Unless a developer readds support for those frequencies. Apparently adding support for MLO (Multi Link Operation a wifi7 feature) required extensive rewrites of existing code and broke the existing 5/10 MHz implementation. Atleast that is what I gather from those 2 commit messages, I could be wrong.

Mediatek/MTK has an open source driver: mt76
But the firmware is not open source. You would probably need to ask them I suppose whether they would consider adding support for smaller channel widths to their firmware. But I don't assume they would do that for an old plattform like ramips mt7621 with mt7905/mt7915 radio. (though I could be totally wrong here) 

their modern plattform running ARM processors is mediatek filogic
so devices like these:
https://github.com/openwrt/openwrt/commit/19d770d7b91aaca1de18281dea0da8616df34c6c

https://github.com/openwrt/openwrt/commit/907cb88e103b2e784caf0803b0ee668da28f2d14
(though 16MiB flash is not good for long term support, you should not buy this device or recommend it)

If you require 5/10 MHz support to stay supported in Linux you might need to reach out to the Linux wireless mailing list and ask kindly whether you can support a developer financially to readd support for this feature else switching to Mediatek won't bear any fruit since you would still be stuck with too old firmware for new devices soon.