aredn/aredn #1670: x86 Generic Image/Proxmox Install Results In Broken Node

StrangeWill, 2024-11-03:
### Describe the bug
x86 Generic Image Does Not Install/Setup Correctly

- [X] Does this issue still occur on the latest version? (**Required**)

(Using the 11/2 nightly build)

### Expected behavior
Installed using the following instructions:

```
root@proxmox-02:/var/lib/vz/template/iso# wget https://downloads.arednmesh.org/snapshots/targets/x86/64/aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img.gz
--2024-11-02 22:45:28--  https://downloads.arednmesh.org/snapshots/targets/x86/64/aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img.gz
Resolving downloads.arednmesh.org (downloads.arednmesh.org)... 45.63.95.75, 2001:19f0:ac00:425b::2
Connecting to downloads.arednmesh.org (downloads.arednmesh.org)|45.63.95.75|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10575788 (10M) [application/gzip]
Saving to: ‘aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img.gz’

aredn-20241102-72655d2d-x86-64-gene 100%[===================================================================>]  10.08M  7.88MB/s    in 1.3s

2024-11-02 22:45:30 (7.88 MB/s) - ‘aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img.gz’ saved [10575788/10575788]

root@proxmox-02:/var/lib/vz/template/iso# gunzip aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img.gz

gzip: aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img.gz: decompression OK, trailing garbage ignored
root@proxmox-02:/var/lib/vz/template/iso# qemu-img resize -f raw aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img 8G
Image resized.
root@proxmox-02:/var/lib/vz/template/iso# qm importdisk 144 ./aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img local-zfs
importing disk './aredn-20241102-72655d2d-x86-64-generic-ext4-combined.img' to VM 144 ...
transferred 0.0 B of 8.0 GiB (0.00%)
transferred 128.6 MiB of 8.0 GiB (1.57%)
transferred 815.1 MiB of 8.0 GiB (9.95%)
transferred 1.5 GiB of 8.0 GiB (18.32%)
transferred 1.6 GiB of 8.0 GiB (19.90%)
transferred 1.8 GiB of 8.0 GiB (21.99%)
transferred 2.5 GiB of 8.0 GiB (30.89%)
transferred 3.1 GiB of 8.0 GiB (39.27%)
transferred 3.8 GiB of 8.0 GiB (47.64%)
transferred 4.5 GiB of 8.0 GiB (56.02%)
transferred 5.2 GiB of 8.0 GiB (64.40%)
transferred 5.8 GiB of 8.0 GiB (72.77%)
transferred 6.5 GiB of 8.0 GiB (81.15%)
transferred 7.2 GiB of 8.0 GiB (89.53%)
transferred 7.8 GiB of 8.0 GiB (97.91%)
transferred 8.0 GiB of 8.0 GiB (100.00%)
transferred 8.0 GiB of 8.0 GiB (100.00%)
Successfully imported disk as 'unused0:local-zfs:vm-144-disk-1'
```

- Upon booting multi-adapter configuration is not detected, putting the system into single adapter
- Upon forcing it via editing `/etc/config/network` things... sort of work
-  `nft` does not properly come online and must be issued `nft insert rule ip fw4 input_wan tcp dport 80` to be able to access the web interface
- `nft` rules must be re-executed on every boot
- NTP time settings do not persist
- Network configuration pop-up fails with the below error:
```
ERROR: left-hand side expression is null
In getDHCP(), file /usr/share/ucode/aredn/configuration.uc, line 209, byte 33:
  called from function main (/app/main/status/e/network.ut:188:41)
  called from function call ([C])
  called from function render ([C])
  called from anonymous function (/app/root.ut:579:18)

 `        const i = iptoarr(setup.lan_ip);`
  Near here -----
```
- Configuration changes do not prompt for restarts
- Probably much more but I've given up on trying to hand-patch this back to a usable state.
- The web UI doesn't agree with the `/etc/config/network` changes I made, forcing it in the UI doesn't fix any of the above (but this does seem to persist in the web UI)

aanon4, 2024-11-03:
Did you follow any set of documented steps? Could you post a complete set of steps you did take, starting from the beginning and not missing anything? I would like to reproduce this but need everything to do that. Thanks.

StrangeWill, 2024-11-03:
The documented steps for a Proxmox install is... sparse at best: https://docs.arednmesh.org/en/latest/arednHow-toGuides/vm-install.html

At this point all it is is an unpacked img and deployed it as a disk image and installing 2 extra NICs that are also paravirtualized (as recommended), then booting the VM. Everything else is kind of off-script since the system boots into an unexpected state (so probably not really applicable since I'm not doing anything that's supported to try to remedy the bad NIC config).

I'm currently reworking some network configs so I can at least get into the web GUI on single-port mode, I'll see if I can at least recover it from there and if the machine starts working as expected.

StrangeWill, 2024-11-03:
[image]

^ Showing it does detect all 3 NICs, but defaults to single-port config

[image]

However I'm still stuck in this state.

There is mention of this issue:

> For example, there is a known issue in the x86_64 firmware before 3.23.12.0 when using more than one Ethernet interface, but this was resolved in subsequent releases.

[image]

But I'm on the nightly build.

aanon4, 2024-11-04:
Can I recommend following these instructions:
https://wiki.bayareamesh.us/index.php/AREDN_on_Proxmox

StrangeWill, 2024-11-04:
> Can I recommend following these instructions: https://wiki.bayareamesh.us/index.php/AREDN_on_Proxmox

Cool, that _mostly_ matched my original, the only different steps:

- I added 2 NICs prior to boot (should be supported)
- I did the first run config on the web GUI (Doing this in the CLI is much nicer but doesn't seem to trigger the NIC discovery)

---

I've tried a few more redeploys with the following changes each:

- Reduced RAM down to 128MB as per requirements (not that this should matter...)
- Removed the 2 additional NICs
- Downloaded the 20241103 nightly
- Didn't expand the image to 8GB, left it as the default 128MB
- Moved from my default CPU type of `x86-64-v2-AES` to just `kvm64` as per your link

The biggest red flag for me is stuff like this with the base image:

```
ERROR: left-hand side expression is null
In getDHCP(), file /usr/share/ucode/aredn/configuration.uc, line 209, byte 33:
  called from function main (/app/main/status/e/network.ut:188:41)
  called from function call ([C])
  called from function render ([C])
  called from anonymous function (/app/root.ut:579:18)

 `        const i = iptoarr(setup.lan_ip);`
  Near here ----------------------^
```

^ showing up in the network configuration, at this point I'm basically a bare system

The only difference I have left is one I have trouble removing: the VLAN tag on the network card, yes this overrides whatever Proxmox is putting out but should still allow me to configure the node correctly.

aanon4, 2024-11-04:
Can you try this again, but using release rather than nightly. Thanks.

StrangeWill, 2024-11-04:
> Can you try this again, but using release rather than nightly. Thanks.

Should have tried this on my own...

So 3.24.10.0 does properly boot and upon first install run _will_ give me a working network config, however it _still_ ignores a multi-adapter setup, keeping eth 1 and 2 unconfigured and down.

I can _probably_ wrangle 3.24.10.0 to eventually start working though FWIW.

aanon4, 2024-11-04:
Right - it's not going to guess how you want multiple network adaptors set up. You need to configure that from the UI once it's up, using the Port dialog you saw earlier. I have production x86 3.24.10.0 nodes on Proxmox with multiple adaptors setup this way. It works fine. Not sure why they're greyed out in your earlier screenshot - it's just reading the status from the kernel and without seeing your Proxmox config I can't really guess what's up with those.

StrangeWill, 2024-11-04:
> Right - it's not going to guess how you want multiple network adaptors set up.

_Hmm_ is the documentation off then?

https://docs.arednmesh.org/en/latest/arednHow-toGuides/vm-install.html

```
Multi-port mode
[...]
In this mode the following ports are automatically assigned:

    First interface: WAN

    Second interface: DtD

    Third and beyond: LAN
```

I'll chase around the adapter issues and report if I find anything weird that isn't somehow caused by my configuration

aanon4, 2024-11-04:
Yes, the documentation is wrong. I'll see about getting that fixed (I thought I had before)

StrangeWill, 2024-11-04:
Cool so mix of nightlies being messed up and incorrect docs, I'm going to assume that the team can reproduce the issue with nightlies too, but if for some reason you can't let me know what I can get you to help gather information about that (I'm not sure where I'd even begin to grab debug data on these issues without some direction).

ab7pa, 2024-11-04:
Updated online documentation to clarify that in multiport mode interfaces are not automatically assigned.  Not sure if anything needs to be updated in the Proxmox wiki instructions.

aanon4, 2024-11-08:
This should be fixed now.

closed 2024-11-08 by aanon4