aredn/aredn #1672: Ubiquity Litebeam 5AC gen 2 - No IP (192.168.1.1) after install

vijaybala1, 2024-11-04:
### Describe the bug
 I installed  3.24.10.0 ,  3.24.6.0 and 20241102 Nightly   for Model 
" Ubiquiti LiteBeam AC Gen2" on a Ubiquity Litebeam 5AC Gen2. The instruction all worked with expected output. After the installation I cannot ping and 192.168.1.1 or get the AREDN Setup page to come up. The MeshNode2G WiFI is visible and if I connected to it I can see that firmware seems to have install correctly and the page @192.168.1.1 comes up. After Setting up the node and rebooting its is still is not accessible over the write 

Setup for install is Laptop with Win11, Dumb TP link switch and Litebeam powered via POE adaptor. 
 
- [ ] Does this issue still occur on the latest version? (**Required**)  Yes. 

### Expected behavior
 After installation, I expect to point the browser at 192.168.1.1 and expect the setup page to come up. 

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
[image]

wu2s, 2024-11-06:
The current firmware works with this device. Try again with the TFTP install and make sure you check you laptop IP address at each step.
[image]

vijaybala1, 2024-11-06:
Ok tried with the latest nightly build from the screenshot above. No luck. Seem like I am doing something incorrectly assuming device is okay.

Setup for install is Windows 11 PC and Device connected to a TP LInk switch with nothing else on the switch, 

 Here are some screenshot and my setups The install progress smoothly . 
1. Reset device  to look TFTP
2. ping 192.168.1.20 .. good
3. transfer 8.7.4 UI firmware .. good
4. bring up UI console. Set the admin password
5. see screenshot - follow instructions to setup
6. wait 10 mins 
7. ping 192.168.1.1 .. no response
8. ping 192.168.1.20  no response
9. Restart device 
10.  ping 192.168.1.1 .. no response
11. ping 192.168.1.20  no response
12. check WiFI - I can see MeshNode2G 
13. connect to MeshNode2G
14.  AREDN page comes up just fine. 

Let me know if anything  can be corrected 

Thanks

[image]
[image]
[image]

aanon4, 2024-11-20:
The thought here is that there has been a minor hardware revision which stops the ethernet working. I have this device, but bought it > 12 months ago, and it works fine. If someone wants to send known broken hardware, that's probably what it will take.