aredn/aredn #1674: UBNT Device Names - Supported Devices List vs. UBNT Catalog

K5RA, 2024-11-04:
List names for 5 GHz AC wireless nodes not exact match to UBNT catalog

[Supported Devices List Product Names Not Matching Vendor Catalog 2024-10-03.docx](https://github.com/user-attachments/files/17248522/Supported.Devices.List.Product.Names.Not.Matching.Vendor.Catalog.2024-10-03.docx)

aanon4, 2024-11-04:
If this is a text file which needs updating, perhaps you could update it and submit a PR? Thanks.

K5RA, 2024-11-05:
Changed AC 5 to 5AC to match catalog and SKU names of Ubiquiti 5AC devices in SUPPORTED_DEVICES.md.

[SUPPORTED_DEVICES.md.5ACedit.txt](https://github.com/user-attachments/files/17631476/SUPPORTED_DEVICES.md.5ACedit.txt)

aanon4, 2024-11-05:
If you are up for it, here's a guide on how to make a Pull Request:
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

closed 2024-12-04 by aanon4