aredn/aredn #1678: Main display fails to appear on GL AR750 after installing NB 20241107

wu2s, 2024-11-07:
### Describe the bug
After installing NB 20241107 on two GL AR750, the whole UI is not visible.
A screenshot would just show a blank page.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-11-07:
What version was this node previously running?

wu2s, 2024-11-07:
Local network issue resolved

closed 2024-11-07 by wu2s

wu2s, 2024-11-07:
The previous day’s build NB 20241106