aredn/aredn #1679: Attached device does not appear in the Local Devices section

wu2s, 2024-11-08:
### Describe the bug
I have a Raspberry Pi web server connected via wired LAN to a Mikrotik hAP AC3. The Pi is connected to port 3 on the hAP. The front panel on the hAP shows port 3 lit up. The Pi has obtained a LAN IP from the hAP as 10.156.233.74. The Ports & Xlinkls section on the hAP shows port 3 active (green background instead of grey).

Even though successfully connected and addressed, there are no devices shown in the Local Devices section on the hAP>

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes - the hAP is running NB 20241107

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
[image]
[image]

K5RA, 2024-11-08:
Do you have to have DHCP reservations and active leases for your Local Devices?
My hAP ac lite shows my RP5B and a VoIP phone, both of which have DHCP entries.

aanon4, 2024-11-08:
Could you re-add the screenshot but dont cut off the LAN DHCP information?

aanon4, 2024-11-08:
Also, please add the support data. Thanks.

wu2s, 2024-11-08:
Yes the hAP shows 1 active lease.
Screenshot and support data are attached.
[supportdata-WU2S-H3-X-2024-11-8-7-51.tar.zip](https://github.com/user-attachments/files/17679620/supportdata-WU2S-H3-X-2024-11-8-7-51.tar.zip)
[image]

K5RA, 2024-11-08:
If you make the DHCP Address Reservation with device name and MAC address, and DoNotPropagate unchecked, that device will appear as a Local Device.  Active lease apparently not required.

wu2s, 2024-11-08:
@K5RA Thank you for the guidance. I determined the MAC address of the Raspberry PI and then went to the hAP to reserve the address. At first, the Pi did not show up in the Local Devices list, even after wait several minutes and doing a screen refresh. Then I went back to the LAN DHCP address reservations and UNCHECKED the Do Not Propagate box as you advised, and the Raspberry Pi instantly appeared in the Local Devices list.

closed 2024-11-08 by wu2s

aanon4, 2024-11-08:
I'm thinking I can do a little better here with what and how this information is generated. I'll take a look.