aredn/aredn #1685: TP-Link CPE510 v2 - Reserved DHCP not propagated

VA2XJM, 2024-11-13:
### Describe the bug
I have two TP-Link CPE510 v2 devices that will not announce the reserved DHCP lease to the mesh. I noticed it tonight, but I believe it is the case since last week when I updated to latest.

I've made the configuration on older version where reservation were announced. Then I updated to latest 3.24.10.0. Today I tried to connect to a device with reservation and hostname is not resolvable from anywhere in the mesh including the node itself.

- [ X ] Does this issue still occur on the latest version? (**Required**) **Yes**

### Expected behavior
Reserved DHCP lease to be announced.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
[supportdata-VA2HRN-FN08LF-CPE-DRI-2024-11-12-20-30.tar.gz](https://github.com/user-attachments/files/17724915/supportdata-VA2HRN-FN08LF-CPE-DRI-2024-11-12-20-30.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
[image]
[image]

### Additional context

aanon4, 2024-11-13:
Will this node announce with a different name? Shorter perhaps? Just curious if you can work out if its something unique about this situation. Not sure I have any nodes announcing devices without associated services.

VA2XJM, 2024-11-13:
Reservation are always announced and this is what makes the hostname reachable outside the local node. 
[image]

Reducing the DHCP reservation hostname does nothing.

I reverted the node to 3.24.6.0 and the hostname of the phone is announced:
[image]

VA2XJM, 2024-11-13:
Ok, I will confirm with the second node but seems that the fact that 3.24.6.0 has been skipped (3.24.4.0 -> 3.24.10.0) is the culprit. 

Having down graded the node to 3.24.6 and re-updated it to latest seems to work. I'll make a check tomorrow to see if the reservation is still announced.

aanon4, 2024-11-13:
If you could send the support data from the node when working that would be helpful. Thanks.

aanon4, 2024-11-13:
I've been unable to reproduce this on the latest nightly. It sounds like there might be an unknown incompatibility between how the data was saved in 3.24.4.0 and now; although 3.24.6.0 wouldnt have changed it unless you 'saved' at some point. It would be useful to see if deleting and recreating the entry in 3.24.10.0 show the problem or not.

VA2XJM, 2024-11-13:
I've tried the delete/save/recreate and no joy. I have a third device at home not being used/deployed that is not yet updated. I'll power it up tonight and see if it the result is the same.

aanon4, 2024-12-04:
Please revisit this with the latest build.

aanon4, 2025-02-10:
No feedback - closing.