aredn/aredn #1689: Mikrotik LHG will not update firmware

wu2s, 2024-11-13:
### Describe the bug
A clear and concise description of what the bug is.
A Mikrotik LHG is running NB 20241111. Attempts to update to the latest NB fail. Tried both normal update via select the available build, then fetch and download, and also download the correct NB from the repository, then upload from the computer to the node.
The fetch and download process initially seems to work in that the fetch appears to succeed, the node goes into its reboot cycle, and then when finished, the node still has the 20241111 firmware on it.
The attempt to upload firmware from the computer is even stranger in that the attempt to select a file from the Downloads folder is not possible because the mouse pointer will not select the Downloads folder when clicking on it. I had to ESC to exit the screen and return to the main page.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes NB 20241111

### Expected behavior
A clear and concise description of what you expected to happen.
Fetch and install firmware successfully like all other versions up to 20241111.
[supportdata-WU2S-LHG-A-2024-11-13-12-12.tar.gz](https://github.com/user-attachments/files/17737493/supportdata-WU2S-LHG-A-2024-11-13-12-12.tar.gz)

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[image]
[image]

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

wu2s, 2024-11-13:
I will install and try another browser. W6BI also reported some difficulties with the Firefox browser.

Here is the WU2S log

[WU2S-syslog.tgz](https://github.com/user-attachments/files/17739658/WU2S-syslog.tgz)

aanon4, 2024-11-14:
Thanks. Unfortunately there wasn't anything useful in it - oh well.
So I've seen this reported before specifically on this hardware, and the way around it is to enabled the "dangerous upgrade" switch. That usually gets the upgrade working, but it doesnt really solve the problem about why this happens sometimes.
Could you download the firmware you wish to upgrade to on your computer, then use that to upgrade the node. After that fails, can you send me that exact firmware file? Thanks.

wu2s, 2024-11-14:
Well, this is interesting, embarrassing and weird.
I just completely emptied my computer's Downloads folder and downloaded the LHG NB 20241113 file (Routerboard-lhg-5hpnd) for the repository.
With only that one file in Downloads, I selected it on the LHG firmware page, followed by a fetch and update.
The node rebooted and this time the firmware update was successful.
I am considering taking videos of all my future updates so that I don't continue questioning my sanity.

closed 2024-11-14 by wu2s

aanon4, 2024-11-14:
If you have this happen again, and you're using the upload procedure, please hang on to the specific firmware file that fails. I wonder if it's being corrupted somehow. I dont know how this could happen, but I'm not sure what else it could be.

wu2s, 2024-11-14:
OK will do.