aredn/aredn #1698: Update Mikrotik hAP AC3 nightly build fails

wu2s, 2024-11-17:
### Describe the bug
A clear and concise description of what the bug is.
Attempted several times to update a Mikrotik hAP AC3 several times from NB 20241115 to NB 20241116 using the normal update procedure (Download firmware, fetch and update). The device fetches and updates, and then displays the Updating screen/Reboot as normal. Upon returning to the main screen after a reboot, the firmware is not updated and remains at NB 2024115. This was tried several times.

Looking at the log on the device I see 3 messages:

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes NB 20241115

### Expected behavior
A clear and concise description of what you expected to happen.

user.info upgrade: The device is supported, but the config is incompatible to the new image (1.0->1.1). Please upgrade without keeping config (sys upgrade -n).

user.info upgrade: Config cannot be migrated from swconfig to DSA

daemon.err https[1259]: Image check failed

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-WU2S-H3-X-2024-11-17-17-23.tar.gz](https://github.com/user-attachments/files/17791698/supportdata-WU2S-H3-X-2024-11-17-17-23.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
Successful update as normal.
[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-11-17:
There was a bug in some nightlies which causes this on some specific devices. To fix type the following once logged in:
```
uci set system.@system[0].compat_version="1.1"
uci commit system
```
You should then be able to upgrade. Let me know if this works.

wu2s, 2024-11-17:
Yes, that did it. Thank you.

closed 2024-11-17 by wu2s