aredn/aredn #1702: Unreliable UI features

Orv, 2024-11-18:
### Describe the bug
On this specific node, the UI feature Reboot only works occasionally, and the Generate Support Feature seems to only work once after a reboot. 

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
Features expected to work as advertised.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
This support data file provided was generated via the command line tool.

[supportdata.tar.gz](https://github.com/user-attachments/files/17805510/supportdata.tar.gz)

aanon4, 2024-11-18:
This node is being constant barraged by tunnel connections which are failing, both incoming and outgoing. Let's get that fixed first then revisit.

Orv, 2024-11-18:
I'll advise the owner.

On 11/18/24 11:42, Tim Wilkinson wrote:
> This node is being constant barraged by tunnel connections which are 
> failing, both incoming and outgoing. Let's get that fixed first then 
> revisit.

Orv, 2024-11-18:
What could be possible causes of that?  His Internet connection looks solid.

Orv, 2024-11-18:
[image]
His tunnels look clean, too.

aanon4, 2024-11-18:
This is a snippet of what the logs are full of:
```
Mon Nov 18 19:33:12 2024 daemon.info vtund[13276]: Denied connection from 70.105.240.237:34805
Mon Nov 18 19:33:12 2024 daemon.err vtund[13277]: Auth failed, unknown host W1YTT-NASHUA1-NH-TUN-172-31-53-180
Mon Nov 18 19:33:12 2024 daemon.info vtund[13277]: Denied connection from 73.4.19.64:39629
Mon Nov 18 19:33:14 2024 daemon.err vtund[13280]: Auth failed, unknown host N2MH-AR-NEWENGLAND-172-31-53-124
Mon Nov 18 19:33:14 2024 daemon.info vtund[13280]: Denied connection from 173.63.130.66:45255
Mon Nov 18 19:33:15 2024 daemon.err vtund[13281]: Auth failed, unknown host N1KWG-AUBURN1-NH-TUN-172-31-53-80
Mon Nov 18 19:33:15 2024 daemon.info vtund[13281]: Denied connection from 73.126.129.182:37423
Mon Nov 18 19:33:15 2024 daemon.err vtund[13282]: Auth failed, unknown host N1MVA-CHESTER2-NH-TUN-172-31-53-84
Mon Nov 18 19:33:15 2024 daemon.info vtund[13282]: Denied connection from 66.31.207.157:40405
Mon Nov 18 19:33:15 2024 daemon.info vtund[6493]: Connecting to meshtunnel.qsoyou.com
Mon Nov 18 19:33:15 2024 daemon.err vtund[6493]: Auth failed, server denied password for host K1EHZ-BEDFORD7-NH-TUN-172-31-152-248
Mon Nov 18 19:33:15 2024 daemon.info vtund[6493]: Connection denied by meshtunnel.qsoyou.com
Mon Nov 18 19:33:16 2024 daemon.err vtund[13309]: Auth failed, unknown host KX1B-GOFFSTOWN1-NH-TUN-172-31-53-172
Mon Nov 18 19:33:16 2024 daemon.info vtund[13309]: Denied connection from 73.119.188.123:40729
Mon Nov 18 19:33:16 2024 daemon.err vtund[13321]: Auth failed, unknown host W1EAA-GOFFSTOWN1-NH-TUN-172-31-53-144
Mon Nov 18 19:33:16 2024 daemon.info vtund[13321]: Denied connection from 71.192.13.254:36517
Mon Nov 18 19:33:17 2024 daemon.err vtund[13330]: Auth failed, unknown host NI1E-GOFFSTOWN1-NH-TUN-172-31-53-168
Mon Nov 18 19:33:17 2024 daemon.info vtund[13330]: Denied connection from 73.69.83.14:42759
Mon Nov 18 19:33:17 2024 daemon.err vtund[13342]: Auth failed, unknown host N1MVA-HAVERHILL1-MA-TUN-172-31-53-156
Mon Nov 18 19:33:17 2024 daemon.info vtund[13342]: Denied connection from 96.70.196.42:41873
Mon Nov 18 19:33:17 2024 daemon.err vtund[13363]: Auth failed, unknown host AC1BG-BROOKLINE1-NH-TUN-172-31-53-184
Mon Nov 18 19:33:17 2024 daemon.info vtund[13363]: Denied connection from 70.105.240.237:46675
Mon Nov 18 19:33:17 2024 daemon.err vtund[13368]: Auth failed, unknown host W1YTT-NASHUA1-NH-TUN-172-31-53-180
Mon Nov 18 19:33:17 2024 daemon.info vtund[13368]: Denied connection from 73.4.19.64:43231
```

k1ky, 2024-11-19:
Are there more than 8 tunnel connections?  Also verify the client Network addys match what is in the host server section.  If a server connection is deleted on the server unit, the network addys may have changed and don't match up?

aanon4, 2024-12-03:
No feedback for 2 weeks - closing

k1ky, 2024-12-03:
I'm late to the party on this one, but sounds very similar to what I am experiencing with Nightly 1202 on HAP and AR-750 units so far.  COmments in "software" on Slack.

aanon4, 2024-12-03:
Please report and provide the data here and then reopen this bug.

k1ky, 2024-12-03:
Will do - I'll wait for the next nightly and test.  I can't seem to be able to obtain a support file under this condition with N-1202.

aanon4, 2024-12-03:
I looked at the thread on Slack, but without a way to reproduce your problem myself and/or the support data for the node, I have no way to help.

k1ky, 2024-12-03:
You should be able to reproduce by disconnecting WAN connection on nodes with Tunnel assignments.  Seems that the instability occurs when there is a Tunnel Client selected but no WAN connection is available.  Any attempt to disable the tunnel client selection seems to aggravate the situation, not able to get support file, won't reboot.  Tried it with N-1203 
and still exists.  Revert to 3.24.10.0 forks fine and stable, able to enable and disable tunnel client connections with no WAN attached.  I can give you access to our system if you need to observe further, but you should be able to duplicate.  AR-750 and HAP ac lite.

k1ky, 2024-12-03:
Attached support file
[supportdata-K1KY-AR750-750-P1-2024-12-3-11-36.tar.gz](https://github.com/user-attachments/files/17996885/supportdata-K1KY-AR750-750-P1-2024-12-3-11-36.tar.gz)

k1ky, 2024-12-03:
HAP Support after reboot
[supportdata-K1KY-HAP-300-P1-2024-12-3-11-47.tar.gz](https://github.com/user-attachments/files/17997018/supportdata-K1KY-HAP-300-P1-2024-12-3-11-47.tar.gz)

k1ky, 2024-12-03:
Once I attempt to disable the tunnel connection, it attempts to commit but doesn't force a reboot.  After that, I can't select reboot or get  support file.

aanon4, 2024-12-03:
So once you're in the state that you cannot select reboot ... tell me how you cannot select reboot as that menu is purely a browser thing; or do you mean the reboot does nothing? Same for support data - the menu is purely browser.
And what can you do in this state? Can you load the device up in a different window? Open other control panels? Can you telnet or ssh into the device?

k1ky, 2024-12-03:
From the browser, when I select reboot or support file, nothing happens - it just seems to hang there.  I'm going to remove MeshChat and see what happens, it may be contributing.

aanon4, 2024-12-03:
If you could answer my other questions above, that would be really useful - thanks.

k1ky, 2024-12-03:
Same results with Firefox.  When I select (push the buttons) reboot or support file, it doesn't appear to react.  Was there another question that I didn't answer?

k1ky, 2024-12-03:
It appears that I can select the hep file and when I go back, it isn't spinning anymore.

aanon4, 2024-12-03:
Once the devices is in this problem state, can you try a few things:
* Can you load the device up in a different browser window? 
* Can you open other control panels?
* Can you telnet or ssh into the device?
* Can you ping the device?

aanon4, 2024-12-03:
> it isn't spinning anymore

The logo is spinning while this is happening? I dont think you had mentioned that.

k1ky, 2024-12-03:
Yes, most of the time the logo is spininng, sometimes not.  I am able to ssh, ping and load in another tab or browser.

aanon4, 2024-12-03:
And can you open panels and generally operate the node?

aanon4, 2024-12-03:
> Yes, most of the time the logo is spininng, sometimes not.

Can you clarify this. I'm going to assume the logo starts spinning when you hit commit after disabling the tunnel(s)? And it keep spinning .. but sometimes stops? When does it stop? How does the node perform once it has stopped and how does this compare to how the node is performing while the node is spinning?

aanon4, 2024-12-03:
Okay ... I think I can replicate this - thanks

k1ky, 2024-12-03:
Great!  The instability occurs at times when the wheel isn't spinning..  I have rebooted my computer, cleared caches and ready to test further if needed.

aanon4, 2024-12-03:
Okay .. this is just weird .. but it looks like any changes to anything is getting the node stuck in a loop internally. Not sure why yet.

aanon4, 2024-12-03:
https://github.com/aredn/aredn/pull/1740
Let's see if this fixes if for you also.

k1ky, 2024-12-04:
N-1204 still hangs on HAP when attempting to change Tunnel Client selection off with no WAN connection.  Seems I was able to turn it on. Will try to reboot and get a support file.

k1ky, 2024-12-04:
Support file attached:

[supportdata-K1KY-HAP-300-P1-2024-12-4-12-7.tar.gz](https://github.com/user-attachments/files/18012443/supportdata-K1KY-HAP-300-P1-2024-12-4-12-7.tar.gz)

aanon4, 2024-12-04:
This support file is not from today's build - looks like it's from 20241203 and not 20241204.

aanon4, 2024-12-04:
Too many support files .. got the right one now - sorry about that.

aanon4, 2024-12-04:
Unlike yesterday, when I could duplicate this issue, today I cannot ... I suppose that's obvious since I checked in a fix that I thought worked.

Rebooting to get me a support file isn't useful I'm afraid, as I need to see the node in the state when it is hung, not after a reboot.

So let's do the following steps. With a tunnel client turned on:
1. Turn off a single tunnel.
2. Before hitting "Commit" note how many changes are pending.
3. Commit .. and then just leave it for 5 minutes (excessive of course, but that's the point).
4. After 5 minutes, assuming it's still spinning, telnet/ssh into the node and type "ps w"
5. Open a new tab/window and see if you can navigate to the node.
6. Assuming you can, try opening and changing the description. We're just trying to see what works at this point and specifically if we can "write" to the node.

Let me know the results of each step. Thanks.

Orv, 2024-12-04:
Remember - we now have a command line support tool :-)

ab7pa, 2024-12-04:
In the docs now...
[image]

k1ky, 2024-12-04:
Here is support file from command line while node is "busy" spinning
[supportdata.tar.gz](https://github.com/user-attachments/files/18013937/supportdata.tar.gz)

k1ky, 2024-12-04:
Results from your test items:
Changed Tunnel Client = ON - 1 commit, no spin
Try to change Client to OFF - 1st try no commit, no spin, 2nd try continuous spin no commit

I think one solution could be to instruct the node to reboot after committing the 1st clean try changes, that way it's in a "clean" state upon return.

I am able to ssh into the node while it is spinning and was able to get a support file (posted above),  but no GUI changes seem to be allowed/saved when in this condition.  I am able to load the node in another TAB, but changes aren't allowed to save and it's still spinning.

aanon4, 2024-12-04:
Can you let me log into this node? I think that's what it's going to take. Thanks.

k1ky, 2024-12-04:
Yes I can log into the node via putty and winscp while it's still spinning.  I posted a support file that I was able to generated from the command line while the node is in this condition

k1ky, 2024-12-04:
> Can you let me log into this node? I think that's what it's going to take. Thanks.

Replied via direct email.

aanon4, 2024-12-04:
Okay.
I've made a couple of changes to this node (and also found another visual bug while I was there - thanks). Can you try it now?

k1ky, 2024-12-04:
Much better - it takes a browser reload after the changes are made, but no reboots are required and I'm able to make changes to the Notes field.  Wasn't giving it much time to respond after the initial changes to see if it stabilizes better.  Was able to turn the Tunnel CLient connections on and off without rebooting.

k1ky, 2024-12-04:
Very STABLE operation with Firefox browser.  Able to turn tunnels on / off within minutes of committing changes and able to edit notes file afterward.

k1ky, 2024-12-05:
I have cycled it ON/OFF twice and updated the Notes file.  A little squirrelly with the display (have to refresh in Chrome) but flawless with FireFox browser.  May be an issue with the number of tabs that I'm actively running in Chrome, but none of the hangup behavior as had been observed before we started all of this before N-1204.  No reboots needed.

Is there anything specific I need to be trying or looking for?

aanon4, 2024-12-05:
Thanks for this.
Nothing specific .. just wanted to make sure my changes didnt break it again .. and seems like they didnt.

Just so it's noted here what's going on:

There is a file lock /tmp/ui.lock which the web server will hold when making a change to the system - any HTTP request which isn't a GET. This is to avoid two changes happening in parallel which could easily cause configurations to be corrupted. For reasons unclear this lock wasn't being released with these tunnel changes despite it not really having anything to do with the tunnel. I think its more network changes in general; although why is still not super clear. Anyway, once the lock is held forever, no other changes are possible and things hang. The way to solve this problem has been to remove the lock, which is how 3.24.10.0 works. But I've been hunting for a way to keep this lock and avoid this problem. This last set of changes (I've been making changes by hand to @k1ky node) seem to finally be a solution to keeping the lock and having the node work.

Tomorrow I'll roll this change out for real and we can all test again.

aanon4, 2024-12-05:
https://github.com/aredn/aredn/pull/1751

k1ky, 2024-12-06:
N-1206 applied.  Toggled Tunnel Client ON/OFF.  All appears to be normal, even in Chrome with screen refresh.

Orv, 2024-12-06:
Loaded nb-1206.  Toggled a disabled tunnel to enabled.  Took about 10 seconds for Commit to clear.   Did notice that hap was unresponsive to pings for approximately 12 seconds then recovered normally.

aanon4, 2024-12-06:
The operations restart the network. Depending on how your connected, it can take while for the network routes to be re-established.

aanon4, 2024-12-06:
But I think we done with this finally.

closed 2024-12-06 by aanon4