aredn/aredn #1705: no longer port forward to the localnode in UI

mathisono, 2024-11-19:
I'm doing a little cleanup and I've depreciated my old rpi GPS PPM ntp server. I have a hap AC2 (20241118-06ed425a) with it's USB gps dongle with gps plug-in installed. 

Im trying to port forward 123 back to the nodes WAN (my home network) 

(new UI 20241118-06ed425a ) I don't see Localnode in the drop down, its not available in port forwarding, only the subnet addresses?

(Old UI 3.24.6.0) I see Localnode in the pull down, as an available option.
 

To conclude I can no longer port forward to the localnode in the UI (new UI 20241118-06ed425a ) 

It may have been missed...? Add this back PLEASE!

aanon4, 2024-11-19:
https://github.com/aredn/aredn/pull/1706

aanon4, 2024-12-02:
@mathisono please verify this fix - thanks.

mathisono, 2024-12-02:
Yes, looks to be responding on the home network, and the localnode appears in the UI.

192.168.3.162   .XFAC.          16 u    -   64    0    0.000    0.000   0.000
 ntp2.net.berkel .GPS.            1 u   54   64    1    5.030  183.243   0.000

closed 2024-12-02 by aanon4