aredn/aredn #1725: Add support for the TP-Link CPE710 v2

jmspswny, 2024-11-28:
### What is the hardware/platform
TP-Link CPE710 v2

### Links
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=5572e0196a93604716ec36ebb8db6737fc6677aa

### Additional context
The V2 of this hardware, which is what is currently available, is now supported in OpenWRT.  Additionally, it appears that the change necessary to support V2 is minimal - the MDIO address of the ethernet port has changed from 0x4 in the V1 hardware to to 0x0 in the V2 hardware.

aanon4, 2024-12-04:
I guess this will be available in the next OpenWRT release; it is not in the current one.

jmspswny, 2024-12-04:
is included in 24.10.x, which is currently in RC.  The most current available image is available here: http://downloads.openwrt.org/releases/24.10.0-rc2/targets/ath79/generic/

jmspswny, 2025-07-25:
as added in PR #1823

closed 2025-07-25 by jmspswny

aanon4, 2025-09-29:
When you say "flash it with OpenWRT" do you mean AREDN? And when you said "through the Web UI" did you not follow these instructions https://docs.arednmesh.org/en/latest/arednGettingStarted/installing_firmware.html#tp-link-first-install-process ?

dkbiz83, 2025-09-29:
My goal is to connect CPE710 to another router so that CPE710 extends the Internet to another location, working in "AP client router" mode (TP-Link's name of the mode). Sorry for a dumb question, but can AREND help me get my goal, and is there a way to flash my Canadian device?