aredn/aredn #1727: Making ANY change within the UI of the AREDN VM kills DTD network

KenArck, 2024-11-30:
### Describe the bug
Editing any parameter from within the UI and then committing, causes the node to loose connection with the host hAP AC2. Sometimes the edit made will survive, sometimes not. 

Checking /etc/config/network shows the ipaddr definition from the dtd section is missing. Adding the line back in (option ipaddr '10.254.69.101' and restarting the network via the CLI (service network restart) restores connectivity.

- [ ] Does this issue still occur on the latest version? (**Required**)
yes, the image I am using is aredn-3.24.10.0-x86-64-generic-ext4-combined.img

### Expected behavior
Changes should be saved and node connectivity should remain

### Support data
Attached

### Screenshots
2 attached, 1 before the problem, 1 after the dtd failure

### Additional context
Add any other context about the problem here.
[supportdata-AH6LE-Proxmox-QTH-2024-11-30-16-18.tar.gz](https://github.com/user-attachments/files/17966058/supportdata-AH6LE-Proxmox-QTH-2024-11-30-16-18.tar.gz)
[image]
[image]

KenArck, 2024-11-30:
Dang network issues

closed 2024-11-30 by KenArck

Orv, 2024-11-30:
Been there, done that...

KenArck, 2024-12-01:
> <https://github.com/aredn/aredn/issues/1727#issuecomment-2509403803>view
> it on GitHub, or

Dunno why I just couldn't get the  3.24.10.0 
release to behave. For grins, I tried the 
3.24.6.0 release but the firsttime_setup script was missing from /usr/bin/local

I now am using the 1128 nightly and it's working perfectly....

------------------------------------------------------------------------------
President and CTO - Arcom Controllers
Makers of repeater controllers and accessories
Phone: (503) 678 6182
https://www.arcomcontrollers.com/
Authorized Dealers for Kenwood and Telewave.
We offer complete turn-key repeater packages!
"We don't just make 'em. We use 'em!"
[]

dpkingston, 2025-01-20:
I am seeing this exact issue on my supernode.  I am not sure the trigger yet, but I will notice the supernode and our tunnel server have lost connection and upon inspection I discover that the supernode has not ip address on its br-dtdlink device.  Looking at /etc/conig/network, I find the stanza for "config interface 'dtdlink'" is lacking an option ipaddr line.  Adding it back and restarting the network restores connectivity.  I do have a /etc/aredn_include/dtdlink.network.user that specifies an alternate ethernet port:
```
root@K7WAN-SEA-SUPERNODE:~#  cat /etc/aredn_include/dtdlink.network.user
# Generated by advancednetwork

config bridge-vlan
	option device 'br0'
	option vlan '2'
	list ports 'eth1:t'

config device
	option name 'br-dtdlink'
	option type 'bridge'
	option macaddr '<dtdlink_mac>'
	list ports 'br0.2'

config interface 'dtdlink'
	option device 'br-dtdlink'
	option proto 'static'
	option ipaddr '<dtdlink_ip>'
	option netmask '255.0.0.0'
```

dpkingston, 2025-01-22:
I switched to nightly build and will see if that makes things more stable.