aredn/aredn #1728: Add ability to do node backups from command line

Orv, 2024-12-01:
A command line-based backup tool would greatly aid the ability to do automated backups of multiple nodes.

Orv, 2024-12-02:
Still testing, but, backup file (and pretty much every file in /tmp) has this datestamp:
root@W6BI-QTH-2G:~# ls -l *.backup
-rw-r--r--    1 root     root          7326 Nov 25 17:27 node-backup.backup

Also, recommend prepending the nodename to the backup file, something like W6BI-QTH-2G.backup.  Would be nice to also have the datestamp, but the node's datestamp isn't guaranteed to be accurate.

Will test restoration accuracy in a bit.

aanon4, 2024-12-02:
The date on the file is from the time on your node .. nothing I can do about that.
Name wise .. no I'm keeping it generic and simple here and let whatever is retrieving it rename it as they seem appropriate. It just makes it harder to automate if the name isn't fixed.

Orv, 2024-12-02:
Good point on the name.  As far as the datestamp goes, here's what it is on the node I generated that backup on:
root@W6BI-QTH-2G:~# date
Mon Dec  2 13:17:15 PST 2024

aanon4, 2024-12-02:
I still dont control the time set on files. Let's do the following:
```
date
cd /tmp
rm -f node-backup.backup
touch time-test-1
/usr/local/bin/backup
touch time-test-2
ls -l /tmp
date
```
And see what it all says

Orv, 2024-12-02:
That time it got the right datestamp:

root@W6BI-QTH-2G:~# cd /tmp
root@W6BI-QTH-2G:~# pwd
/tmp
root@W6BI-QTH-2G:~# touch time-test-1
root@W6BI-QTH-2G:~# /usr/local/bin/backup
Generated backup file: /tmp/node-backup.backup
root@W6BI-QTH-2G:~# touch time-test-2
root@W6BI-QTH-2G:~# ls -l /tmp
-rw-r--r--    1 root     root            23 Nov 25 17:26 TZ
-rw-r--r--    1 root     root          1088 Nov 25 17:25 ath9k-eeprom-ahb-18100000.wmac.bin
-rw-r--r--    1 root     root             0 Nov 25 17:28 dhcp.leases
drwxr-xr-x    2 root     root            60 Nov 25 17:28 dnsmasq.d
drwxr-xr-x    3 root     root           100 Nov 25 17:26 etc
drwxr-xr-x    2 root     root            60 Nov 25 17:26 hosts
drwxr-xr-x    3 root     root            60 Nov 25 17:26 lib
drwxrwxrwt    2 root     root           380 Dec  2 13:11 lock
drwxr-xr-x    2 root     root            80 Nov 25 17:25 log
-rw-r--r--    1 root     root          5163 Dec  2 13:53 lqm.info
-rw-r--r--    1 root     root            18 Dec  2 13:38 lqm.phy0.macs
-rw-r--r--    1 root     root             0 Dec  2 13:53 metrics-ran
-rw-r--r--    1 root     root          7326 Dec  2 13:53 node-backup.backup
-rw-r--r--    1 root     root           158 Dec  2 13:54 node.history
-rw-r--r--    1 root     root            11 Dec  2 13:54 olsrd.watchdog
drwxr-xr-x    2 root     root            40 Nov 25 17:26 opkg-lists
drwxr-xr-x    2 root     root            40 Dec 31  1969 overlay
-rw-r--r--    1 root     root            21 Nov 25 17:28 resolv.conf
lrwxrwxrwx    1 root     root            35 Nov 25 17:28 resolv.conf.auto -> /tmp/resolv.conf.d/resolv.conf.auto
drwxr-xr-x    2 root     root            60 Nov 25 17:26 resolv.conf.d
drwxr-xr-x    7 root     root           360 Nov 25 17:26 run
-rw-r--r--    1 root     root           242 Nov 25 17:28 service-validation-state
drwxrwxrwt    2 root     root            40 Dec 31  1969 shm
-rw-r--r--    1 root     root            57 Dec  2 13:53 snr.dat
drwxr-xr-x    2 root     root            80 Dec  2 13:01 snrlog
drwxr-xr-x    2 root     root            60 Nov 25 17:26 state
drwxr-xr-x    2 root     root           120 Nov 25 17:26 sysinfo
-rw-r--r--    1 root     root             0 Dec  2 13:53 time-test-1
-rw-r--r--    1 root     root             0 Dec  2 13:54 time-test-2
-rw-r--r--    1 root     root             3 Dec  2 12:59 timesync
drwxr-xr-x    2 root     root            40 Nov 25 17:25 tmp
-rw-r--r--    1 root     root             0 Nov 25 17:26 ui.lock
drwxr-xr-x    3 root     root            60 Nov 25 17:26 usr
drwxr-xr-x    2 root     root            80 Nov 25 17:26 vtun
-rw-r--r--    1 root     root           877 Dec  2 13:53 wireless_monitor.info
root@W6BI-QTH-2G:~# date
Mon Dec  2 13:54:19 PST 2024

aanon4, 2024-12-02:
Thanks. Closing as complete.

closed 2024-12-02 by aanon4