aredn/aredn #1732: DHCP LAN reservations don't save

KenArck, 2024-12-01:
### Describe the bug
Creating LAN reservations by clicking on the plus sign (next to current leases) and then clicking DONE closes the popup window but the usual banner about COMMIT never shows. Re-opening the LAN DHCP window shows nothing has changed. See the BEFORE and AFTER screenshots

- [ ] Does this issue still occur on the latest version? (**Required**)
This is the latest nighly 20241128-d7754d89

### Expected behavior
I expect the COMMIT banner to be displayed and the added reservations to be saved
[image]
[image]

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

KenArck, 2024-12-01:
[supportdata-AH6LE-ProxMoxNode-QTH-2024-12-1-19-40.tar.gz](https://github.com/user-attachments/files/17970748/supportdata-AH6LE-ProxMoxNode-QTH-2024-12-1-19-40.tar.gz)

KenArck, 2024-12-02:
> <https://github.com/aredn/aredn/issues/1732#issuecomment-2510313481>view
> it on GitHub, or

I'm batting 1000 this week...sheesh....

------------------------------------------------------------------------------
President and CTO - Arcom Controllers
Makers of repeater controllers and accessories
Phone: (503) 678 6182
https://www.arcomcontrollers.com/
Authorized Dealers for Kenwood and Telewave.
We offer complete turn-key repeater packages!
"We don't just make 'em. We use 'em!"
[]

KenArck, 2024-12-02:
Not sure if this has anything to do with it but I also notice the AREDN logo doesn't spin when you click the Node Status icon to reload the page

aanon4, 2024-12-02:
See above - fixed tomorrow

KenArck, 2024-12-02:
Are x86 Proxmox nodes updated via the usual mechanism?

aanon4, 2024-12-02:
Please verify this fix - thanks

KenArck, 2024-12-02:
Verified - reservations are now correctly saved, thanks

closed 2024-12-02 by aanon4