aredn/aredn #1737: Node Status Local Devices have Link Icons

K5RA, 2024-12-03:
System: hAP-ac-lite with 2024-12-02 NB, RP5B with latest bookworm and Chromium.  See same effect seen with the hap-ac-lite and  Win 11 PC with Firefox 133.0.

When I upgraded from 3.24.10.0, my two Local Devices  show link icons  

[image]
[image]

to right of device name on the Node Status page. One is a local RP5B, other is a local VoIP phone. On Mesh Status page, no icon is associated with either local device.

Expected behavior:   link icon is not displayed in Node Status display with Local Devices that are not links.

[supportdata-K5RA-HAP-FB-2024-12-2-21-8.tar.gz](https://github.com/user-attachments/files/17986982/supportdata-K5RA-HAP-FB-2024-12-2-21-8.tar.gz)

aanon4, 2024-12-03:
What happened when you clicked on them?

K5RA, 2024-12-03:
In Node Status screen: 
    Clicking on the rp5b1 device brings up Apach2 Debian Default Page.
    Clicking on VoIP-fb brings up Grandstream 1625 log-in page..

aanon4, 2024-12-03:
So this is doing exactly what it is now designed to do. The local device section will now detect if the device has a http(s) page it can open, and will turn the device name into a link if so. These auto-generated links are *not* published services so will not be published on the mesh page. They are convenience links if you are using the node directly.

This was a feature request. Glad it works.

closed 2024-12-03 by aanon4