aredn/aredn #1746: Deleted Wireguard tunnel still reported in logs

Orv, 2024-12-04:
### Describe the bug
A clear and concise description of what the bug is.
A Wireguard tunnel was deleted via the UI several weeks ago.   However I find this recurring in the logs:

2024-09-20T15:59:51-07:00 W6BI-SHACK-NODE wireguard_monitor: wgs0 endpoint k6pvr.ham-radio-op.net:5525 is not responding for 528 seconds, trying to re-resolve hostname

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
The node should have forgotten about the tunnel upon its deletion

### Support data
[supportdata-W6BI-SHACK-NODE-2024-12-4-11-32.tar.gz](https://github.com/user-attachments/files/18013395/supportdata-W6BI-SHACK-NODE-2024-12-4-11-32.tar.gz)

Orv, 2024-12-04:
Disregard - filed in error

closed 2024-12-04 by Orv

aanon4, 2024-12-04:
I was going to say .. I dont see this error in your logs