aredn/aredn #1754: DHCP leases not displayed

KenArck, 2024-12-08:
### Describe the bug
Latest nightly (20241207-cbfbad9d) running x86 Proxmox. Working DHCP leases are not displayed, only reservations are

- [ ] Does this issue still occur on the latest version? (**Required**)
Nightly 20241207-cbfbad9d

### Expected behavior
Expect to see all active DHCP leases displayed

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.
[supportdata-AH6LE-ProxMoxNode-QTH-2024-12-7-18-53.tar.gz](https://github.com/user-attachments/files/18050565/supportdata-AH6LE-ProxMoxNode-QTH-2024-12-7-18-53.tar.gz)
[image]

### Additional context
Add any other context about the problem here.

KenArck, 2024-12-08:
[image]
[image]

aanon4, 2024-12-08:
How long did you wait? Leases will only repopulate as the devices renew them. I dont see this on my nodes.

KenArck, 2024-12-08:
I waited 20 minutes or so. I came back to it later and they're displayed. Sorry for the false flag.
However, I do notice the AREDN logo no longer spins when refreshing the page (for example, clicking the !). Should I start a new issue or ignore it?

aanon4, 2024-12-08:
It only spins if the update takes more than a certain period of time. I never see it spin on proxmox because the underlying machine is usually fast enough. This hasn't changed.

closed 2024-12-08 by aanon4

KenArck, 2024-12-08:
OK. Yea, I can't believe how slow the hAP AC2 seems to have been in comparison to this one.