aredn/aredn #1761: MikroTik hAP ac3 - Web UI not working - Nightly

JPurp, 2024-12-11:
### Describe the bug
After applying the nightly build the Web UI won't load. It page shows loading. 

- [ ] Does this issue still occur on the latest version? (**Required**) Yes

### Expected behavior
The Web UI should load properly after power off. 

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
No UI access to pull support data. 

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
The node Web UI worked with 3.24.10 but node need to upgrade to nightly build to address a dtd networking bug. After upgrading to nightly the node works properly until shutdown. When powered back on the Web UI won't load. The node is issuing DHCP on LAN and I can SSH into the node.  This issue was originally observed on 12/6 nightly. I also applied the 12/9 nightly with the same issue.

aanon4, 2024-12-11:
As you can ssh in, please use `/usr/local/bin/supportdata` to generate the support data - thanks.

aanon4, 2024-12-11:
I loaded nightly onto my ac3. UI was fine. Power cycled it. UI was fine when it came back up.
So perhaps something specific about your configuration? - so I will definitely need that support data to do more.

JPurp, 2024-12-11:
Here is the support file.

[supportdata.tar.gz](https://github.com/user-attachments/files/18089494/supportdata.tar.gz)

aanon4, 2024-12-11:
Well the web server is running and there are no errors in the log. Can you provide more details about what the web browser says when it fails to render the UI? Does it work if you use an IP address rather than a hostname? If you use 'curl' from the command line to get the page, what do you see?

aanon4, 2024-12-11:
Also ... how did you upgrade once you had loaded a nightly without a working UI?

JPurp, 2024-12-11:
Attempting to connect via IP gives nothing;
[image]

Connecting via localnode.local.mesh resolves to the node host name and  gets stuck with the spinning wheel;
[image]

I updated from 12/6 to 12/9 nightly by PXE booting to the in memory image and then applying the 12/9 nightly.  

-JP

aanon4, 2024-12-11:
Did you have chance to try access with curl? Really I'm just looking for something less intelligent to see if the web server responds correctly.
How does it behave if you access it with Firefox (I assume that's Chrome in the pics)?

JPurp, 2024-12-11:
I tested with Firefox and same result. I have curl response from both IP and hostname;
[image]

I broke the first rule of troubleshooting by changing the environment but I think I found the bug.

I updated to the 12/11 nightly via CLI and then made changes one at a time doing a UI reboot and then a power off reboot at each step;

Wizard - OK
Set location - OK
Change WiFI from WAN to LAN, set SSID and enable WPA2 - FAIL

I made the commit change and the UI was still working. Once I selected reboot it failed.

-JP

aanon4, 2024-12-11:
https://github.com/aredn/aredn/pull/1764

JPurp, 2024-12-11:
Great news. Is this a quick fix that would show up in a nightly? I am open to testing. 

Thanks,
-JP

aanon4, 2024-12-11:
It will in nightly in the morning for testing.

JPurp, 2024-12-12:
12/12 Nightly installed, tested and web UI is working including a power off reboot.

I am seeing what I think is another bug with LAN WiFi. Here is what I am seeing;

WPA2 - SSID not broadcast. Can't connect to network (tried hidden).
WPA - SSID not broadcast. Can't connect to network.
None - Working

Is this intended operation?

Thanks. -JP

aanon4, 2024-12-12:
Sounds like a different bug. Please open a new issue for it. Thanks.

closed 2024-12-12 by aanon4