aredn/aredn #1763: Ubiquiti Rocket 5 AC Lite

jjkroell, 2024-12-11:
### Describe the bug
AREDN firmware flashes but when at the 192.168.1.1/a/status 'Welcome page', I enter the Node Name (as directed) and Password then click 'Save & Reboot'.
I am presented with the Rebooting page with the IP address of the node and the five minute countdown.
After the five minutes (I have waited over night as well), the node is not at the new AREDN IP address but back at 192.168.1.1
I have tried both static and dynamic IP addresses, have tried using Windows 10, Windows 11 and Mint Linux OS on three separate PC's.

- [ ] Does this issue still occur on the latest version? 
Yes (even on nightly).

### Expected behavior
Once the five minute countdown is finished the node should have been rebooted and have AREDN firmware loaded and is accessible with localnode.mesh or the MAC based IP address. 

### Support data
Unable to provide Support data from the latest firmware however I managed to load 3.23.12 and downloaded the support data here:
https://www.arednmesh.org/sites/default/files/sites/default/files/pictures/user-VE7KOD/supportdata-NOCALL-202304272031.tgz

### Screenshots
Here is a video of my install from start to finish 

https://youtu.be/QgkcHxSyGOI

https://www.arednmesh.org/sites/default/files/sites/default/files/pictures/user-VE7KOD/IMG_1761.jpg

### Additional context
Add any other context about the problem here.

aanon4, 2024-12-11:
Unless you have a really good reason, I would not suggest installing a nightly build from scratch at this time.  Because installing from scratch is a very time consuming and manual process to QA, we only do this towards the end of a release cycle and we're not near that at the moment. I would suggest installing the latest release build and upgrading to nightly if running nightly is your goal (and I've installed the latest release on Rocket 5AC Lites many times without issue).

That said, thanks for taking the time to report this and I will take a look at the install process later this week.

jjkroell, 2024-12-11:
I was grasping at straws trying to trouble shoot what was going on. I tried every single version I could get my hands on to see if I could get it over this speed bump.

Very willing to send the hardware to someone if that is needed at some point (once the Canada Post situation gets sorted).

Thanks for the quick reply.

aanon4, 2024-12-11:
Can you try with this specific factory install?
https://downloads.arednmesh.org/releases/3/24/3.24.10.0/targets/ath79/generic/aredn-3.24.10.0-ath79-generic-ubnt_rocket-5ac-lite-squashfs-factory.bin

jjkroell, 2024-12-11:
Done.
I temporarily had a 'loading AREDN' page (have never seen this), then the Welcome page at 192.168.1.1/a/status
But still having the same issue unfortunately, not at 10.164.68.216 and still at 192.168.1.1 with the initial Welcome page.

aanon4, 2024-12-11:
If you're comfortable using ssh or telnet then perhaps you can get me some more information? If so:
* Flash the unit with the latest nightly build
* When you get to the Welcome page, dont do anything there. Instead ...
* Login via telnet or ssh. The address will be 192.168.1.1, username root and password hsmm
* Run `/usr/local/bin/supportdata`
* This will generate a file in /tmp (it will tell you its exact name). Copy this file off the node (scp from your host machine) and send it to me.

Thanks. I'm hoping the boot up logging will say something useful.

jjkroell, 2024-12-11:
With the .bin file you had me try last night 3.24.10.0

`/usr/local/bin/supportdata : not found`

Here is a ls of `/usr/local/bin/`

`aredn_message.sh        linkled                           recoverymode
aredn_postupgrade      manager.lua                  restart-services.sh
aredn_sysupgrade        mesh-firewall                setpasswd
firmwarecheck.sh         metrics                          spawn_sysupgrade
firstuse-setup              mgr                                upgrade_kill_prep
get_boardid                node-setup                     upgrade_prepare.sh
get_hardware_mfg     nvram-setup                   uploadctlservices
get_hardwaretype      olsrd-namechange          usb_passthrough
get_model                  poe_passthrough            wscan`

aanon4, 2024-12-11:
Please note the first line of the instruction required the nightly build:

```Flash the unit with the latest nightly build```

jjkroell, 2024-12-12:
Sorry! Glad I put the version up front so we weren't chasing our tails!

Received error: `cp: can't stat '/tmp/dnsmasq.d//*': No such file or directory` 
Here is the support data file that was generated after receiving the above error.

[supportdata.tar.gz](https://github.com/user-attachments/files/18103764/supportdata.tar.gz)

aanon4, 2024-12-12:
Thanks.
So the logs are filled with messages like this:
```
[  144.269219] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b3378
[  144.286795] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b3c74
[  144.296869] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b3e34
[  144.305859] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b3f28
[  144.315413] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b4050
[  144.324838] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b4180
[  144.334274] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b42ac
[  144.343719] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b43d4
[  144.360387] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b4b30
[  144.378085] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b5420
[  144.408423] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b6d64
[  144.448434] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b9398
[  144.457714] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b947c
[  144.473845] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009b9ac4
[  144.490189] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009ba128
[  144.506901] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009ba874
[  144.523025] jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 009baedc
```
Which dont seem good. Is this brand new hardware? I wonder if Ubiquiti have done a hardware revision - they almost never note it when they do.

jjkroell, 2024-12-12:
I purchased this off Ebay December 2023 and having issues flashing AREDN since I received it, definitely not brand new. I can crack the case and take photos if you think that may help.

aanon4, 2024-12-12:
Dont do that - wont mean anything to me.
Do you have any other Rocket AC lites to try?

jjkroell, 2024-12-12:
No unfortunately this was a single unit.

aanon4, 2024-12-12:
Looks like the flash chips in your hardware is different to the ones in mine (I checked a few I have in services).

Mine: mx25l12805d
Yours: en25qh128

I will need to do some research to see if this flash chip has known issues (some of them do)

jjkroell, 2024-12-12:
Thanks for your help Tim.

aanon4, 2024-12-12:
In reality I suspect I'll need the thing on my desk. If you want to email me your address I'll ship you a replacement and you can send me this one.

jjkroell, 2024-12-12:
This is not operational, why don't I just send it to you? No need to swap it out but I appreciate the offer. I'll email you though.

Jesse

aanon4, 2024-12-19:
For the radio in the mail today - thanks.
Yes, the kernel code for EON flash chips (which is what this uses) if just broken. The flash chips used in this device is essentially identical to the Macronix one use in my devices - two suppliers, two chips, drop in compatible - but the fixes for the Macronix chips were not ported over. Once that's done, this device works correctly.

I'll roll the fix into the pending OpenWRT 24.10 merge as the old and new flash code is sufficiently different that I dont want to write this fix twice.

Thanks for sending this. Hope you received the replacement in the mail already. I'd like to hang on to this because OpenWRT will break this again as it's not being fixed upstream in the main Linux codebase.

jjkroell, 2024-12-19:
> Received the radio in the mail today - thanks.

You didn’t waste any time getting to work on that!

Of course you can keep that unit, appreciate all the development work you
and the team have put in to the AREDN project.
You weren’t charged anything were you (taxes/duties etc)?

Wasn’t aware you had already shipped the replacement up. Canada Post was
just ordered back to work yesterday so they will have quite the backlog to
get through.

Thanks again for all your help!

Jesse

closed 2025-02-02 by aanon4