aredn/aredn #1766: MikroTik hAP ac3 - LAN WiFI Encryption Not Working

JPurp, 2024-12-12:
### Describe the bug
MikroTik hAP ac3 LAN WiFI SSID not broadcast when WPA/WPA2 configured.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Support data

### Additional context
MikroTik hAP ac3 is running 12/12 nightly to address dtd networking issues and web UI bug (1761). When configuring LAN WiFi I see the following;

WPA2 - SSID not broadcast. Can't connect to network (also tried hidden).
WPA - SSID not broadcast. Can't connect to network.
None - Working

I have tested different unlicensed channels (1, 6, 11) as well as changing SSID and done soft/power off reboots. Result is consistent.

aanon4, 2024-12-12:
Looks like this support data is for 'none'. Could you add one for 'wpa' or 'wpa2' ?

aanon4, 2024-12-12:
Also please make sure the password is valid (I think at least 8 characters) as we dont check this in the UI at the moment. Finally, if you dont see the network then please do a reboot and check again in case there's a problem there.

JPurp, 2024-12-12:
Resolved

The issue was minimum password length. I left the default AREDN password and tried other there were <8. Once a proper password it came right up.

aanon4, 2024-12-13:
https://github.com/aredn/aredn/pull/1767

Thanks for confirming.

closed 2024-12-19 by aanon4