aredn/aredn #1771: Port Forwarding ranges and minor UI clean up

jjkroell, 2024-12-19:
### Describe the bug
- Not able to add a range of ports in the LAN port input box but can in the WAN input box.  If I want to forward a range of 10 ports, should they not go to their corresponding internal ports and not to a single port. I know other firewalls you can leave the LAN port box empty and it matches what you input in the WAN box.

- When adding a third (or more Port Forward, the 'minus/delete' button is out of place (see screenshot)

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
If I want to port forward a range of external ports they should go to a range of internal ports, not just a single internal port.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-VE7KOD-AP-2024-12-18-21-22.tar.gz](https://github.com/user-attachments/files/18192933/supportdata-VE7KOD-AP-2024-12-18-21-22.tar.gz)

### Screenshots

[image]

### Additional context
Add any other context about the problem here.

aanon4, 2024-12-19:
Look at the port forwarding section here for how the WAN and LAN port ranges work.

https://docs.arednmesh.org/en/latest/arednGettingStarted/node_admin.html#local-services

jjkroell, 2024-12-19:
OK.. sorry - disregard the range thing..  apologies!

aanon4, 2024-12-22:
https://github.com/aredn/aredn/pull/1775

closed 2025-01-29 by aanon4